motioneye-project/motioneye #983: MotionEye Server/Hub without motion deamon?

Bootscreen, 2018-08-22:
Hello,

i have 3 PI's as Fast Network Cameras and a FreeBSD Server which runs MotionEye to show me all the streams in one place.
I don't need streaming, motion detection, image or movie capturing on the server, the only thing i currently use is the overlay.

is it in this case possible to disable the motion deamon on the server? when i comment the path to the binary out, motioneye can't find it itself and in the web interface i get this line: Motion Version | None
thats what i wan't no ressource hungry motion deamon. but when i do this, the web interface don't show me the live images from my cameras :(

so is it possible to use it without the motion deamon or has anyone an other solution for me?

Best regards
Bootscreen

ccrisan, 2018-08-22:
If your cameras are added as Remote motionEye Cameras, then you should be able to see them. I assume you have added them as Network Cameras, which make use of the local motion.

ccrisan, 2018-08-22:
Yes, again, if you add your cameras as Remote motionEye Cameras to your hub.

Bootscreen, 2018-08-22:
Ok, again. I don't want the motion daemon anywhere. When i add them as
Remote motionEye camera, i have a slow pi cam and Not the nice fast network
cam.

But I understand. Motioneye can't Do this.

Does anyone know an other Software which can act as a hub?

nonsintetic, 2018-08-22:
If all you need is to view the streams on one page, why not make a HTML web-page with iframes for each camera feed. With fast camera you get a feed that shows the camera video on port 8081 (you get a link from your fast camera settings under "Streaming"). 
You can embed those feeds on a single web page, no need for another Motioneye install just to see them all in one place.

Bootscreen, 2018-08-22:
because why i should reinvent the wheel when i have a nice solution? a web
page with login and responsive design for my smartphone, tablet and pc
costs time. and motioneye has all of this ;)

Bootscreen, 2018-08-23:
> @Bootscreen <https://github.com/Bootscreen>, you have then the FNC mode on
> slaves + Simple MJPEG Camera on the Hub. No motion there, but you'll have
> to consider ports 808x if you need external access.

Thanks, that is what i'm looking for :D

closed 2018-08-23 by Bootscreen