motioneye-project/motioneye #985: Cannot save pictures nor movies.

WoodyKu, 2018-08-26:
I installed MotionEye 0.39.3 on Raspberry Pi3 running Raspbian Stretch. I followed the installation as defined in: https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
Everything seems to be working fine except: 
- making manual pictures does not create any pictures in the Camera1 sub-folder. 
- motion detection does not create pictures, either.
- no movies saved.

ccrisan, 2018-08-26:
Please attach screenshots of your configuration menu so we can see how you set up your system.

WoodyKu, 2018-08-26:
Here are the configuration screenshots.
[pics.zip](https://github.com/ccrisan/motioneye/files/2321934/pics.zip)

ccrisan, 2018-08-28:
Your system appears to be properly set up for what you want. Please attach your motion.log file as well as the motioneye log (usually obtainable using journalctl -u motioneye).

WoodyKu, 2018-08-31:
I could not find motion.log, so I attach the output form journal and syslog.
[syslog.txt.gz](https://github.com/ccrisan/motioneye/files/2341833/syslog.txt.gz)
[motioneye.journal.gz](https://github.com/ccrisan/motioneye/files/2341834/motioneye.journal.gz)

JamBalaya56562, 2026-11-08:
Closing, as the attached journal shows the cause: Motion fails with `VIDIOC_S_INPUT: Device or resource busy` and "Error capturing first image" every few seconds, and motionEye keeps restarting it. Another program already had the camera open, so Motion never got a picture to save. If pictures are still not saved with nothing else using the camera and a current release, please open a new issue with the log.

closed 2026-11-08 by JamBalaya56562