motioneye-project/motioneye #986: Where do i set the time zone?

joland, 2018-08-28:
I searched all the internet but no definite answer on how to set the time zone. Current (probably by default) timezone is UTC. 
[root@meye-61c67703 etc]# date
Tue Aug 28 11:29:38 UTC 2018

But i need CET.

I am running this version:

motionEye Version | 0.39.3
Motion Version | 4.1.1
OS Version | motionEyeOS dev20180730

FYI, i haven't seen an option to change the timezone in the UI either...

ccrisan, 2018-08-28:
Have you enabled advanced settings?

joland, 2018-08-28:
I managed to change with : date -s
However it is not a permanent solution. After reboot it loses the changes.
If anyone knows how to do a permanent fix, let me know.

ccrisan, 2018-08-28:
Let's see a screenshot of your settings panel with various sections opened.

joland, 2018-08-28:
@ccrisan I am sorry and can't believe I completely looked over this. 
Admin -> General Settings -> Time zone.

closed 2018-08-28 by ccrisan

ITCMD, 2020-02-11:
I just made the same mistake and after a few days found your post! THANKS!