motioneye-project/motioneye #987: Xiaofang (Fang-Hacks)

ageorgios, 2018-08-28:
There is a very popular camera Xiaofang (and cheap)
there is a hack for it that installs an open source rtsp server 
snx_rtsp_server -W 1920 -H 1080 -Q 5 -b 4096 -a

but the motioneye is not cooperating with this snx rtsp server correctly (it crashes the server)

ccrisan, 2018-08-28:
Please describe your scenario in more detail. Are you trying to add your RTSP camera to motionEye? If yes, what does the `snx...` command have to do with motionEye/

JamBalaya56562, 2026-11-09:
The details asked for did not come, and there was no news since 2018, so closing. If the camera's RTSP server still crashes when added on a current version, please open a new issue with the camera URL type and the motion log.

closed 2026-11-09 by JamBalaya56562