motioneye-project/motioneye #989: motioneye.service failed to start

judalvarezca, 2018-08-29:
Hello! First of all, i have been enjoying motioneye for a while, it's excellent! Thank you for all the hard work.

I'm working with 3 RPi Zero W built-in Cameras (Raspbian+Motioneye), under a virtualenv python shell, and a server with access from a public IP address. All was going good, but suddenly one of my cameras stopped working. 

Debugging the problem, I checked the status for my services, obtaining the following:

`pi@raspberrypi:~ $ systemctl status
● raspberrypi
    State: degraded
     Jobs: 0 queued
   Failed: 1 units
    Since: Wed 1969-12-31 19:00:01 -05; 48 years 7 months ago
   CGroup: /
           ├─user.slice
           │ └─user-1000.slice
           │   ├─user@1000.service
           │   │ ├─gvfs-gphoto2-volume-monitor.service
           │   │ │ └─696 /usr/lib/gvfs/gvfs-gphoto2-volume-monitor
           │   │ ├─dbus.service
           │   │ │ └─459 /usr/bin/dbus-daemon --session --address=systemd: --nof
           │   │ ├─gvfs-udisks2-volume-monitor.service
           │   │ │ └─678 /usr/lib/gvfs/gvfs-udisks2-volume-monitor
           │   │ ├─gvfs-mtp-volume-monitor.service
           │   │ │ └─704 /usr/lib/gvfs/gvfs-mtp-volume-monitor
           │   │ ├─gvfs-goa-volume-monitor.service
           │   │ │ └─700 /usr/lib/gvfs/gvfs-goa-volume-monitor
           │   │ ├─gvfs-afc-volume-monitor.service
           │   │ │ └─691 /usr/lib/gvfs/gvfs-afc-volume-monitor
           │   │ ├─init.scope
           │   │ │ ├─442 /lib/systemd/systemd --user

pi@raspberrypi:~ $ systemctl --failed
  UNIT              LOAD   ACTIVE SUB    DESCRIPTION                            
● motioneye.service loaded failed failed motionEye Server                       

LOAD   = Reflects whether the unit definition was properly loaded.
ACTIVE = The high-level unit activation state, i.e. generalization of SUB.
SUB    = The low-level unit activation state, values depend on unit type.

1 loaded units listed. Pass --all to see loaded but inactive units, too.
To show all installed unit files use 'systemctl list-unit-files'.`

I checked my SD for error and all seems good. No clue of how to work around this, and if it's not related at all to motioneye. Here is my service setup:
`[Unit]
Description=motionEye Server

[Service]
ExecStart=/home/pi/.virtualenvs/mymotioneye/bin/meyectl startserver -c /home/pi/motioneye/conf/motioneye.conf
Restart=on-abort

[Install]
WantedBy=multi-user.target`

Thank you in advance!

chuskywoj, 2018-09-10:
Hi ! I Have similar issue. What I see is that `pip install motioneye` dosent copy any files to `/usr/local/bin`. Simply **meyectl** is missing in that location.

That was tested on the fresh latest Raspbian. Running **meyectl** directly from Python directory works well and gives video streaming. 

Any thoughts what is going wrong with missing files in  `/usr/local/bin` ??

judalvarezca, 2018-09-10:
What do you see after ```which meyectl```?

chuskywoj, 2018-09-11:
```
pi@raspberrypi:~ $ which meyectl
pi@raspberrypi:~ $ meyectl
-bash: meyectl: command not found
pi@raspberrypi:~ $ meyectl
```

chuskywoj, 2018-09-11:
ok now it works. The reason mostly was that I missed `sudo` when installing with `pip`.
I had to uninstall motion eye with pip and manual removal of remaining folders. Then by installing again with `sudo pip` works :) 
I think it could be better if there would be error notification during installation in case of **missing sudo**. Most probably this cause that some files are not copied.

judalvarezca, 2018-10-01:
Well, I already resolve my problem, so I'm closing the issue for now.

closed 2018-10-01 by judalvarezca

binitshah, 2021-06-03:
Just ran into the same issue described by @chuskywoj. In the [Raspbian setup guide](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian), `pip install motioneye` silently fails to copy to motioneye files to `/usr/local/share/...`, and reports a successful installation when the python parts succeed.

It'd be great if the commands had "sudo" in front of them, or a note was added to the [troubleshooting](https://github.com/ccrisan/motioneye/wiki/Troubleshooting) section.

starbasessd, 2021-06-03:
@binitshah 
https://github.com/ccrisan/motioneye/wiki/Installation
note 1: The given commands normally need to be run as root; type them in a root shell or use sudo before each command.
-----
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
Before Proceeding

    Read the general Installation page first.
    These instructions apply only to an up-to-date Raspbian Stretch.
    All commands require root; use sudo before each command or become root using sudo -i.
-----
The instructions are there.