motioneye-project/motioneye #994: Feature Request - Access Point for configuring Wifi

Jose-Monteiro, 2018-08-31:
https://davesteele.github.io/comitup/

I was starting to develop a gui in gambas but found the package above. The idea behind it: boot up, test for a connection; if fails, create an AP with a pre-determined IP and offers a web portal for connection with discovered AP's. If connection succeeds, proceed booting the system.

It could be extremely useful, mainly to establish Wifi networking on a headless device.
Could it be done?

ccrisan, 2018-09-02:
What we currently have is https://github.com/ccrisan/motioneyeos/wiki/WiFi-Access-Point. Long story short: you need to manually add some config files before it works.

jasaw, 2018-09-03:
See https://github.com/ccrisan/motioneyeos/issues/1422

JamBalaya56562, 2026-11-09:
This is about motionEyeOS, which has its own repository; as linked above, the access point is covered there (wiki and motioneyeos#1422). Nothing to change in motionEye itself, so closing.

closed 2026-11-09 by JamBalaya56562