motioneye-project/motioneye #995: Streaming URL not available in docker.

aptonline, 2018-09-01:
I've successfully setup motioneye in Docker on my Synology but one thing that doesn't seem to work in the streaming url.  The port produced is obvious local to the docker container and not the host system (`8081`).  Is there anyway to produce a URL scheme similar to the still url (which works).

RichJackson, 2018-09-28:
have you tried running the docker container with the
`--network host`
flag?

kleini, 2018-10-02:
At least the port forward into the container for 8081 is missing in the Docker description in the Wiki. Will add that. I am not sure about the streaming URL.

JamBalaya56562, 2026-11-09:
As noted, port 8081 needs to be published as well; the current `docker/docker-compose.yml` does that (`8081:8081`), or `--network host` works. Then the streaming URL works with the host's address. Closing.