motioneye-project/motioneye #997: Motioneye instable

viczena, 2018-09-07:
Motioneyes works for some minutes after boot, then one or more of the 4 cameras start to dissappear. "cant open webstream".

Sometimes the stream recovers, sometimes not. If i let Motioneyes run for several hours, it completly stops working. It is still logged in to the wifi router, but cannot be accessed anymore.

This is not a product, it is just a half-baked prototype of something. To bad I spent the time trying to relove issues.

sfgabe, 2018-09-21:
No one ever accused github of hosting fully formed products. :) 

That said, I've also experienced this issue with the latest version. I've been using motionEye and motioneyeos for projects for quite a while now and it's always been pretty reliable, so it seems like a new issue. 

It's difficult for me to troubleshoot because it is so intermittent - as soon as I reboot everything's fine for a few hours and then I get the grey X camera instead of a stream. It stops taking still images as well.

ccrisan, 2018-09-21:
@viczena you did not even bother to find out what the actual name of the "non-product" you've been using is; "motioneyes" does not exist. If you use "motionEyeOS", you're in the wrong place. If you use "motionEye" on top of some Linux distro, it may just as well be the fault of the underlying OS.

Either way, you omitted to attach any significant logs or to specify anything relevant to your setup.

ccrisan, 2018-09-21:
@sfgabe have you experienced this issue with both motionEye and motionEyeOS? Let's try to isolate the problem and see if we can find the root cause of it.

ccrisan, 2018-09-21:
@sfgabe there weren't any recent significant changes in motionEye so your problem might be caused by something else. If the problem still occurs, take a look over the following logs:

 * `journalctl -u motioneye`
 * `/var/log/motion.log`

Have you also recently updated Motion?

closed 2018-09-21 by ccrisan

sfgabe, 2018-09-21:
@ccrisan any clues? Usually I can decipher & troubleshoot but "unexpected camera control callback event" is unfamiliar to me.

journalctl -u motioneye:
`Sep 21 09:17:39 octocam motion[389]: [1:ml1] [ERR] [VID] mmalcam_next: cmd 0 flags 00000004 size 0/1536000 at aed98
Sep 21 09:17:39 octocam motion[389]: [0:vc.ril.camera] [ERR] [VID] Received unexpected camera control callback even
Sep 21 09:17:39 octocam motion[389]: [1:ml1] [ERR] [VID] mmalcam_next: cmd 0 flags 00000004 size 0/1536000 at af088
Sep 21 09:17:39 octocam motion[389]: [0:vc.ril.camera] [ERR] [VID] Received unexpected camera control callback even
Sep 21 09:17:39 octocam motion[389]: [1:ml1] [ERR] [VID] mmalcam_next: cmd 0 flags 00000004 size 0/1536000 at aef10
Sep 21 09:17:39 octocam motion[389]: [0:vc.ril.camera] [ERR] [VID] Received unexpected camera control callback even
Sep 21 09:17:54 octocam meyectl[222]:    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
Sep 21 09:18:04 octocam meyectl[222]:    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
Sep 21 09:18:08 octocam motion[389]: [0:motion] [ERR] [ALL] main: Thread 1 - Watchdog timeout, trying to do a grace
Sep 21 09:18:14 octocam meyectl[222]:    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
Sep 21 09:18:24 octocam meyectl[222]:    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
`
/var/log/motion.log
`[0:motion] [NTC] [ALL] motion_startup: Motion 4.0.1 Started
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1:ml1] [ALR] [VID] mmalcam_start: MMAL Camera thread starting... for camera (vc.ril.camera) of 1280 x 800 at 10 f$
[1:ml1] [ERR] [VID] mmalcam_next: cmd 0 flags 00000004 size 0/1536000 at aed98020
[0:vc.ril.camera] [ERR] [VID] Received unexpected camera control callback event, 0x0004c614
[1:ml1] [ERR] [VID] mmalcam_next: cmd 0 flags 00000004 size 0/1536000 at af088020
[0:vc.ril.camera] [ERR] [VID] Received unexpected camera control callback event, 0x0004c614
[1:ml1] [ERR] [VID] mmalcam_next: cmd 0 flags 00000004 size 0/1536000 at aef10020
[0:vc.ril.camera] [ERR] [VID] Received unexpected camera control callback event, 0x0004c614
[0:motion] [ERR] [ALL] main: Thread 1 - Watchdog timeout, trying to do a graceful restart
[0:motion] [ERR] [ALL] main: Thread 1 - Watchdog timeout, did NOT restart graceful,killing it!
[0:motion] [ERR] [ALL] main: cleaning Thread 1
[0:motion] [ALR] [VID] MMAL Camera cleanup
`

sfgabe, 2018-09-27:
To follow up on this... it looked like my camera was dying so I replaced it and swapped the board also just to make sure. Got a great new picture! ...but still having the motioneye stall problems. Currently it runs for about 20 hours and then stalls with the following error. 

> [1:ml1] [ERR] [VID] mmalcam_next: cmd 0 flags 00000004 size 0/1536000 at af010020
> [1:ml1] [ERR] [VID] mmalcam_next: cmd 0 flags 00000004 size 0/1536000 at aee98020
> [1:ml1] [ERR] [VID] mmalcam_next: cmd 0 flags 00000004 size 0/1536000 at af010020
> [0:vc.ril.camera] [ERR] [VID] Received unexpected camera control callback event, 0x0004c614
> [0:vc.ril.camera] [ERR] [VID] Received unexpected camera control callback event, 0x0004c614
> [0:vc.ril.camera] [ERR] [VID] Received unexpected camera control callback event, 0x0004c614
> [0:motion] [ERR] [ALL] main: Thread 1 - Watchdog timeout, trying to do a graceful restart
> [0:motion] [ERR] [ALL] main: Thread 1 - Watchdog timeout, did NOT restart graceful,killing it!

I tried restarting motion and motioneye separately but it doesn't make a difference, the entire board needs to be rebooted to get the camera back. I'm not sure if this points to raspbian, motion, or motioneye as the problem, but I suspect it's a conflict between those. At the moment I'm just rebooting once a day with a cronjob but if there is a better solution lease let me know!

motion 4.0.1
motioneye 0.3.93
raspbian 9.4

ccrisan, 2018-09-29:
@sfgabe the "unexpected camera control callback event" is something that when googled for, shows up associated with various hardware issues. I understand you replaced your camera already. Can you maybe try replacing other components (even the board itself)?

motionEye is at the top of the stack and it's very unlikely to be the cause of your problem here.