motioneye-project/motioneye #999: Stop/Start continuously recording  with good connection state

menschmachine-de, 2018-09-08:
Hi,

motioneye keep recording continuously even when a cam is lost its connection.
It is possible to implement continuously recording only when a camera is in good connection state?
Best regards

ccrisan, 2018-09-08:
That's what [Motion](https://github.com/Motion-Project/motion) does - it records missing video information as gray frames.

closed 2018-09-08 by ccrisan

menschmachine-de, 2018-09-10:
Hi ccrisan,

thx for your quick answer.
 
Your right its "not**your**problem" ..but i have to ask again.
I contact the motion developer.

He told me that its possible to do this with scripts for  'on_camera_lost' and 'on_camera_found'

```

!#/bin/bash

curl http://{yourip}:{port}/{threadid}/config/set?emulate_motion=on

Sample script/documentation:
https://rawgit.com/Motion-Project/motion/master/motion_config.html#OptDetail_Webcontrol
https://motion-project.github.io/motion_config.html#on_camera_lost
```

It is possible to use/implement this in motioneye?

Best regards

ccrisan, 2018-09-12:
I understand that could be done with some workarounds by switching off and on the recording by hooking to those events but:

1. This wouldn't be easy to implement and would add unnecessary complexity to the code and configuration.
2. I believe this should be handled by Motion itself (maybe using a configuration flag) - so that frontends won't have to manually switch on and off the recording when you practically have no video feed.