motioneye-project/motioneye #1001: Timestamps

rohtua, 2018-09-14:
Hi, I've been running a motioneye server for a week or so now and I've noticed that the timestamp is an hour out and when I change the timezone it has no effect at all it always sits an hour behind the current time.

jasaw, 2018-09-17:
What's your system time on your motioneye server? Is it also 1 hour behind?
Are you syncing time via NTP?

IVData, 2018-10-02:
Having the same issue on Docker. Time is always UTC, even thought I have /etc/localtime and /etc/timezone passed through.

IVData, 2018-10-02:
Uncommenting this line in motioneye.conf causes Docker container to not start:

```
# path to the localtime file
# (enable this to configure the system time zone from the UI)
#local_time_file /etc/localtime
```

```
INFO: hello! this is motionEye server 0.39.3
Traceback (most recent call last):
File "/usr/local/bin/meyectl", line 11, in <module>
load_entry_point('motioneye==0.39.3', 'console_scripts', 'meyectl')()
File "/usr/local/lib/python2.7/dist-packages/motioneye/meyectl.py", line 255, in main
server.main(arg_parser, sys.argv[2:], command[:-6])
File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 437, in main

File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 362, in run
make_media_folders()
File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 296, in make_media_folders
config.get_main() # just to have main config already loaded
File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 185, in get_main
_get_additional_config(main_config)
File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 2112, in _get_additional_config
(sections, configs) = get_additional_structure(camera=bool(camera_id))
File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 2083, in get_additional_structure
result = func()
File "/usr/local/lib/python2.7/dist-packages/motioneye/tzctl.py", line 126, in timeZone
import pytz
ImportError: No module named pytz
```

kleini, 2018-10-02:
Please provide the output of the following commands

    $ zdump /etc/localtime
    /etc/localtime  Tue Oct  2 22:04:05 2018 CEST
    $ docker exec -it motioneye zdump /etc/localtime
    /etc/localtime  Tue Oct  2 22:02:32 2018 CEST

IVData, 2018-10-03:
I solved this by adding the apt package `python-tz` at build time. This seems to be required for any timezone functionality at all. With only that change my timezone was set correctly without even passing `/etc/localtime` but my system (Unraid) does pass the `TZ` environment variable by default.

Can put through a pull request if you want.

kleini, 2018-10-03:
Within the publicly available Docker image python-tz is not necessary. Therefore I am not sure, why it is required in your image?

kleini, 2018-10-03:
The import in tzctl.py line 126 is there since 2015. How can I trigger motioneye to enter that part of the code? My motioneye installation does not show the above error and therefore I never stumbled across that exception.

IVData, 2018-10-03:
You can see above 126 that if `LOCAL_TIME_FILE` is defined, the import line will be run. I had uncommented the line `#local_time_file /etc/localtime` in motioneye.conf in an attempt to get my timezone working properly.

It turns out that there are a few problems with the docker container:
1. If the user defines `local_time_file` in config, the container crashes. pytz is required.
2. Setting a timezone in the docker container does not work by any method - I think this is because of the Ubuntu package tzdata missing.

So, installing python-tz and tzdata fixes both of these issues.

kleini, 2018-10-04:
@rohtua @jasaw @IVData Can you please test the Docker image tagged with timestamps-amd64 or timestamps-armhf whether they fix your issue?

IVData, 2018-10-06:
Working great for me =)

kleini, 2018-10-07:
Fixed with next release of motioneye.

closed 2018-10-07 by kleini