motioneye-project/motioneye #1003: playback/timelapse by phone and windows unavailable for motioneye created files

davebdb, 2018-09-18:
I followed the directions several times for Ubuntu (running Lubuntu flavor of Ubuntu, to lighten the resource usage).  

I tried both the native install and also built ffmpeg from source and also motion from source.

"preview" playback and timelapse work fine while using my linux based laptop, but when I try to play back or timelapse from a windows PC or my cellphone I get an error.  

when doing a "timelapse" with phone (android running chrome) I get the following error:
It starts out with 1% then fails:
"The timelapse movie could not be created"

when doing a movie preview:
"Error:  Media decode error or unsupported media features"

When using a motioneyeOS (Pi 3B version) all these features have no issues and work as planned.  Since I want to put more than 1 or 2 cameras up and want to be able to run them with HD support I need to put them on a full size server. 

Is there some library or configure option I am missing or need to enable/disable?

davebdb, 2018-09-18:
Additional info:
I forgot to add, that the video files are perfectly fine if I download and play them with the local player such as VLC.  So the files themselves seem to be intact even for the failed system, but must be downloaded instead of "streamed" off the server.  It would be nice to preview it before downloading.

Klausoleum, 2018-10-18:
I have a very similar problem.
[1022](https://github.com/ccrisan/motioneye/issues/1022)

On a Linux (Ubuntu) laptop, the timelapse function only works sometimes.
Via Terminal I can always create timelapse videos

With debian and other distributions I have the same problems
With motioneyeOS it works flawlessly

ccrisan, 2018-10-19:
This is probably due to a video codec that's not supported by the browser. Make sure you're running ffmpeg and motion versions equal or close to those used by motionEyeOS and that ffmpeg is compiled with the codecs you're using.

kunau, 2018-12-09:
Running these versions:
motionEye Version | 0.39.3
Motion Version | 4.1.1
OS Version | motionEyeOS 20181129

Movie Format H.264 (.mp4)

With the same issues. Tracking appears to work but produces a movie that is not playable on any of my platforms, LINUX, IOS, or OSX. When a movie is selected in the browser is produces the error:

**Error: Media format unsupported or otherwise unavilable/unsuitable for playing.**

(sic, including the mis-spelling of unavailable.)

Will continue to monitor this thread and will respond if I can offer a fix.

Thanks,

andresanches, 2019-01-29:
@davebdb I had a similar issue when trying to view recorded footage (stream, not download) on my android phone (on chrome). I noticed that the playback _almost_ started with the browser actually showing recording length in minutes:seconds but then crapped out. I also noticed that on an actual computer streaming stuttered a lot because buffering was slower than the playback speed- strange because both the server and the client computer are on a wired gigabit network. I suspected it was a bottleneck in the code causing a timeout and as it turns out, at least for me, it was

Looking [at the code](https://github.com/ccrisan/motioneye/blob/28a663e8fa774e5c2f9aa4337affdf3fa879a9b9/motioneye/handlers.py#L1608) I found that the `MoviePlaybackHandler` uses the `/tmp` directory heavily, which in my case was on the filesystem. I then mounted the tmp directory as `tmpfs` instead, according to [this suggestion](https://unix.stackexchange.com/a/220851)

and now not only playback works fine on android but the stuttering on the computer is also gone

Hope this helps others who run into this issue

Janmaniek, 2019-05-22:
I appear to have the exact same issue.

_Please bear in mind that I'm not a programmer or well versed or capable in altering anything from command lines, or reading log files. I'm often dependent on clear instructions that others have shared in places such as Github(thank you all)._

On my Android phone(Chrome) I cannot playback video files from one of my two devices, it returns the error message: Error: "**Media decode error or unsupported media features**." These same video files can be watched on other devices, such as my Chromebook(Chrome). On the same phone I can view videos recorded on my other(dedicated) motionEyeOS device.

My setup is as follows:

Raspberry Pi 3B+ connected by wifi (my dedicated device that has no other function).
Operating as intended. recorded video can by viewed on any device from the web interface.

motionEye Version | 0.40
Motion Version | 4.2.2
OS Version | motionEyeOS 20190427

-----------

Raspberry Pi 3B+ wired connection
Primary purpose is running Pi-hole(this functions as intended)

motionEye Version | 0.40
Motion Version | 4.2.2
OS Version | Raspbian 9.9 (stretch lite)

I installed motionEye in this device last weekend following the instructions in the guide on this site and it worked very well. A day later I noticed Pi-hole had a new version available, I updated to it. The webinterface of motionEye from this point on showed the new Raspbian OS version ( was 9.8, now is 9.9) and here is when I stopped being able to watch videos on my phone. 
Throughout all of this the version of motion remained 4.1.1. In an effort to fix the issue, I did update to 4.2.2 manually, but motion does not seem to be causing the problem. I fear I messed something up by updating and now the codecs or some other dependency or something related to ffmpeg don't work right.

Not only that, the videos that are recorded also open very slowly on devices that can open them, even when downloaded, much as andresanches described. Changing movie formats doesn't make a difference, I've changed it back to the default H.264/OXM which worked before.
I tried following the instructions that andresanches pointed to in his last post, but I don't know if I did it properly, if it was the solution I needed and it has not helped me so far(but it might if done correctly).

ffmpeg -version returns:
ffmpeg version 3.2.12-1~deb9u1+rpt1 Copyright (c) 2000-2018 the FFmpeg developers
built with gcc 6.3.0 (Raspbian 6.3.0-18+rpi1+deb9u1) 20170516

The entirety of my motioneye.log is this:

-- Logs begin at Thu 2016-11-03 18:16:42 CET, end at Wed 2019-05-22 17:46:12 CEST. --
May 22 15:01:20 raspberrypi systemd[1]: Started motionEye Server.
May 22 15:01:23 raspberrypi meyectl[443]:     INFO: hello! this is motionEye server 0.40
May 22 15:01:26 raspberrypi motion[565]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
May 22 15:01:26 raspberrypi motion[565]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
May 22 15:01:26 raspberrypi motion[565]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
May 22 15:01:26 raspberrypi motion[565]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.2.2 Started
May 22 15:01:26 raspberrypi motion[565]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
May 22 15:01:26 raspberrypi motion[565]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
May 22 15:01:28 raspberrypi meyectl[443]:     INFO: cleanup started
May 22 15:01:28 raspberrypi meyectl[443]:     INFO: wsswitch started
May 22 15:01:28 raspberrypi meyectl[443]:     INFO: tasks started
May 22 15:01:28 raspberrypi meyectl[443]:     INFO: mjpg client garbage collector started
May 22 15:01:28 raspberrypi meyectl[443]:     INFO: server started
May 22 15:01:39 raspberrypi motion[565]: [1:ml1:Buiten] [**WRN] [ENC] ffmpeg_set_codec: Preferred codec h264_omx has been blacklisted**

That last line in particular seems like a big clue to me, but I wouldn't know what to do about it and after some googling, I can't find any solution to it.

I feel like there is an easy fix to this and that I will slap myself in the face once I learn about it, but so far I'm out of luck. I hope someone here knows what to do. I'll gladly supply anyone with more details if that's necessary. 

Thanks and keep up the good work, in the short time that I've tinkered with motionEye I'm very impressed by it.

Technaur, 2019-07-26:
I am also getting the error message (While streaming on the web interface)
 **Error: Media format unsupported or otherwise unavilable/unsuitable for playing.**

I am running versions:
motionEye Version | 0.40
Motion Version | 4.1.1
OS Version | Raspbian 10

When I use to run motionEyeOS it worked fine. 
I did also set up a custom path for my motioneye.conf file in directory /home/pi/motioneye/motioneye.conf

Is some codec for streaming on the web interface missing or something?

With setup I did follow this guide.
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
and 
https://github.com/ccrisan/motioneye/wiki/Running-As-Normal-User

davebdb, 2019-12-31:
@Technaur can you check to see if this solution works for you as well?  It did for me
https://github.com/ccrisan/motioneye/issues/1067#issuecomment-443541808

I'm going to post the output of ffprobe on the 2 files.  One that doesn't work (pre convert.sh script implementation) and one that works (after the script was added.  They look the same to me, so not sure what the issue is exactly.  Maybe someone can spot something that is different between the 2, other than file size because they are not the same file as they are different times.

DOES NOT WORK:
$ ffprobe -i 10-19-48.mp4 -show_format -hide_banner -show_streams
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '10-19-48.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf57.83.100
  Duration: 00:00:15.63, start: 0.036000, bitrate: 20640 kb/s
    Stream #0:0(und): Video: h264 (High 4:4:4 Predictive) (avc1 / 0x31637661), yuv420p, 1280x720, 20637 kb/s, 23.60 fps, 24 tbr, 16k tbn, 48 tbc (default)
    Metadata:
      handler_name    : VideoHandler
[STREAM]
index=0
codec_name=h264
codec_long_name=H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10
profile=High 4:4:4 Predictive
codec_type=video
codec_time_base=1737/82000
codec_tag_string=avc1
codec_tag=0x31637661
width=1280
height=720
coded_width=1280
coded_height=720
has_b_frames=0
sample_aspect_ratio=N/A
display_aspect_ratio=N/A
pix_fmt=yuv420p
level=31
color_range=unknown
color_space=unknown
color_transfer=unknown
color_primaries=unknown
chroma_location=left
field_order=unknown
timecode=N/A
refs=1
is_avc=true
nal_length_size=4
id=N/A
r_frame_rate=24/1
avg_frame_rate=41000/1737
time_base=1/16000
start_pts=576
start_time=0.036000
duration_ts=250704
duration=15.669000
bit_rate=20637517
max_bit_rate=N/A
bits_per_raw_sample=8
nb_frames=369
nb_read_frames=N/A
nb_read_packets=N/A
DISPOSITION:default=1
DISPOSITION:dub=0
DISPOSITION:original=0
DISPOSITION:comment=0
DISPOSITION:lyrics=0
DISPOSITION:karaoke=0
DISPOSITION:forced=0
DISPOSITION:hearing_impaired=0
DISPOSITION:visual_impaired=0
DISPOSITION:clean_effects=0
DISPOSITION:attached_pic=0
DISPOSITION:timed_thumbnails=0
TAG:language=und
TAG:handler_name=VideoHandler
[/STREAM]
[FORMAT]
filename=10-19-48.mp4
nb_streams=1
nb_programs=0
format_name=mov,mp4,m4a,3gp,3g2,mj2
format_long_name=QuickTime / MOV
start_time=0.036000
duration=15.633000
size=40334212
bit_rate=20640548
probe_score=100
TAG:major_brand=isom
TAG:minor_version=512
TAG:compatible_brands=isomiso2avc1mp41
TAG:encoder=Lavf57.83.100
[/FORMAT]

WORKS:
$ ffprobe -i 16-06-50.mp4 -show_format -hide_banner -show_streams
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '16-06-50.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf57.83.100
  Duration: 00:00:12.64, start: 0.000000, bitrate: 1660 kb/s
    Stream #0:0(und): Video: h264 (High) (avc1 / 0x31637661), yuv420p, 1280x720, 1657 kb/s, 25 fps, 25 tbr, 12800 tbn, 50 tbc (default)
    Metadata:
      handler_name    : VideoHandler
[STREAM]
index=0
codec_name=h264
codec_long_name=H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10
profile=High
codec_type=video
codec_time_base=1/50
codec_tag_string=avc1
codec_tag=0x31637661
width=1280
height=720
coded_width=1280
coded_height=720
has_b_frames=2
sample_aspect_ratio=N/A
display_aspect_ratio=N/A
pix_fmt=yuv420p
level=31
color_range=unknown
color_space=unknown
color_transfer=unknown
color_primaries=unknown
chroma_location=left
field_order=unknown
timecode=N/A
refs=1
is_avc=true
nal_length_size=4
id=N/A
r_frame_rate=25/1
avg_frame_rate=25/1
time_base=1/12800
start_pts=0
start_time=0.000000
duration_ts=161792
duration=12.640000
bit_rate=1657942
max_bit_rate=N/A
bits_per_raw_sample=8
nb_frames=316
nb_read_frames=N/A
nb_read_packets=N/A
DISPOSITION:default=1
DISPOSITION:dub=0
DISPOSITION:original=0
DISPOSITION:comment=0
DISPOSITION:lyrics=0
DISPOSITION:karaoke=0
DISPOSITION:forced=0
DISPOSITION:hearing_impaired=0
DISPOSITION:visual_impaired=0
DISPOSITION:clean_effects=0
DISPOSITION:attached_pic=0
DISPOSITION:timed_thumbnails=0
TAG:language=und
TAG:handler_name=VideoHandler
[/STREAM]
[FORMAT]
filename=16-06-50.mp4
nb_streams=1
nb_programs=0
format_name=mov,mp4,m4a,3gp,3g2,mj2
format_long_name=QuickTime / MOV
start_time=0.000000
duration=12.640000
size=2622868
bit_rate=1660043
probe_score=100
TAG:major_brand=isom
TAG:minor_version=512
TAG:compatible_brands=isomiso2avc1mp41
TAG:encoder=Lavf57.83.100
[/FORMAT]

wx-watch, 2020-01-10:
Your convert script helped me in an unrelated fashion. I could not play files from my RTSP camera on my Mac or iPhone and would get the "Error: Media decode error or unsupported media features" error. It worked on a Linux box. But I discovered that Quicktime can't play the High 4:4:4 Predictive profile that the RTSP camera was delivering. 

So I forced the profile by adding "-profile:v high" to the ffmpeg command, and now they play fine!

davebdb, 2020-07-05:
@ccrisan question on how this can be resolved.  I've noticed the thing that is different is the profile for the streams that have issues is the predictive 4:4:4 

NOT WORK
Stream #0:0(und): Video: h264 (High 4:4:4 Predictive) (avc1 / 0x31637661), yuv420p, 1280x720, 20637 kb/s, 23.60 fps, 24 tbr, 16k tbn, 48 tbc (default)
WORKS
Stream #0:0(und): Video: h264 (High) (avc1 / 0x31637661), yuv420p, 1280x720, 1657 kb/s, 25 fps, 25 tbr, 12800 tbn, 50 tbc (default)

JamBalaya56562, 2026-11-07:
Closing as a duplicate of #1206: the files that fail use the `High 4:4:4 Predictive` profile, as you found above, which many browsers cannot play. Motion writes such files at 100% movie quality, so a lower quality gives files that play in the browser.

closed 2026-11-07 by JamBalaya56562