motioneye-project/motioneye #1006: Motion Eye frezzing and locking up on Pi 3B+

alphanumeric007, 2018-09-22:
I have motion Eye OS running on two Pi3B's with no issues. However on my brand new 3B+ it locks up and freezes. I lose the feed and have to power cycle the Pi. I've rebooted my PC but it still won't connect. Usually the video just freezes, and the web UI doesn't respond. If I close my browser and try to connect it won't connect. I've reimaged and switched Micro SD cards. WIFI connection on the 2.4 GHz band as it will not connect on the 5Ghz band. It will connect on the 5GHz band in Raspbian no problem, just not in Motion Eye. Version 2 Pi camera on the CSI.
motioneye Version 0.39.2
motion version 4.1.1
OS version motioneyeos 20180627

ccrisan, 2018-09-29:
This might be a duplicate of https://github.com/ccrisan/motioneyeos/issues/1661. Please provide logs with debugging enabled.

JamBalaya56562, 2026-11-08:
Closing, as ccrisan pointed to motioneye-project/motioneyeos#1661 and the debug logs he asked for never came. This was on motionEyeOS, which is maintained in its own repository, so please follow up there if it still happens.

closed 2026-11-08 by JamBalaya56562