motioneye-project/motioneye #1009: Camera Select and Camera Remove Button defaulting to display none

hurricanepkt, 2018-09-29:
I have :  

RaspberryPi 3 running motionEyeOS 20180627 with MotionEye Version 0.39.2 and Motion Version 4.1.1
x86 server running Ubuntu 17.10 with MotionEye Version 0.39.3 and Motion Version 4.0

Today when accessing from chrome, firefox and ie from multiple computers. After I've logged in as the admin user.

The select element id="cameraSelect" and the div id="remCameraButton" both have style of display:none

``` html 
<div class="settings-top-bar closed">
	<div class="button icon settings-button mouse-effect" title="settings"></div><div class="button icon logout-button mouse-effect" title="switch user"></div>
	<select class="styled" id="cameraSelect" style="display: none;"><option value="3">Camera3</option><option value="add">add camera...</option></select>
	<div class="button icon rem-camera-button mouse-effect" id="remCameraButton" title="remove camera" style="display: none;"></div>
	<div class="button apply-button" id="applyButton" style="opacity: 0; display: none;"><img class="apply-progress" src="static/img/apply-progress.gif"></div>
	<div class="hostname">hostname</div>
</div>
```

This prevents adding or removing cameras

ccrisan, 2018-09-30:
Have you enabled Fast Network Camera?

hurricanepkt, 2018-10-01:
I have on the Pi 3 with direct cameras but the Ubuntu connects to my cameras over the network

ccrisan, 2018-10-02:
I understand you have two systems:

 * a Raspberry PI running motionEyeOS
 * an Ubuntu machine running motionEye

Which of the two were you accessing from Chrome when you saw the missing add/remove camera functionality?

hurricanepkt, 2018-10-03:
Both machines,

ccrisan, 2018-10-03:
@hurricanepkt 

1. Have you enabled Fast Network Camera on your Raspberry PI machine?
2. Have you changed the `add_remove_cameras` setting in your motioneye.conf file?