motioneye-project/motioneye #1010: when making a MotionEye Server?

shadowfxd, 2018-09-30:
Does GPU matter for the server? I'll be using Pi 2's as IP cameras running Motioneyeos, then I'm planning on using a intel NUC running Ubuntu but does GPU matter or CPU only? What about RAM?

ccrisan, 2018-09-30:
GPU won't make any difference - most of the heavy lifting is done by CPU. RAM, on the other hand, could be the bottle neck in some cases.

shadowfxd, 2018-09-30:
is 4GB RAM good or 8GB at the minimum?

ccrisan, 2018-09-30:
@shadowfxd it really depends on what you're doing with your motionEye. But I'd say 4GB is more than enough.

JamBalaya56562, 2026-11-07:
Closing as answered above: most of the work is done by the CPU, RAM can be the limit, and 4 GB is more than enough in most setups.

closed 2026-11-07 by JamBalaya56562