motioneye-project/motioneye #1013: Thumb

nickilo, 2018-10-02:
Hello, 

First i would like to congratulate about your nice piece of software.

I'm encountering a problem with the thumb jpg file created after a motion detection. I would like the thumb to contain the motion, and if perfect the start or the middle of this motion, but sometime it does not show the motion at all.

I have read the source file, and trying to correct myself, but i should miss something.

What do you need to reproduce, i can give you sample video et jpg thumb and configuration files.

If needed i can even give you temporary access to the serveur.

Regards.

Nickilo

richri, 2018-11-02:
I think this is a duplicate request as in Issue #814 - I am searching for exactly the same ...

JamBalaya56562, 2026-11-09:
Closing as a duplicate of #814, which asks for the same: a thumbnail that shows the motion. Please follow and vote there.

closed 2026-11-09 by JamBalaya56562