motioneye-project/motioneye #1014: Video playback/download slow on motionEye+Raspbian compared to motionEyeOS

jazeph, 2018-10-03:
First of all I want to tell that I am very impressed with this very cool software.

I am currently experimenting with motionEye on Raspbian Stretch and motionEyeOS and noticed that video playback on motionEye+Raspbian is much slower than on motionEyeOS. I don't have exact numbers, but playback on motionEyeOs is just smooth as if the video is played locally, and playback on motionEye+Raspbian is stuttering all the time.

The Raspberry Pi 3B has a wired network connection to the router. I uninstalled as many as possible packages from Raspbian Stretch to be sure I don't have unnecessary stuff running. I did install OpenCV because I want to get a notification only when a motion is triggered by a detected person, but I haven't activated any of this yet.

Can anybody point me to a clue on where I could investigate this? Thanks in advance!

ccrisan, 2018-10-03:
@jazeph there's no common/known reason for a difference in performance between the two setups. What  versions are you running on the two setups (motion, ffmpeg, motionEye)?

jasaw, 2018-10-04:
@jazeph If you're recording the video locally on the Pi, then most probably your motionEye + Raspbian setup hasn't enabled GPU hardware acceleration for video encoding.
See https://github.com/ccrisan/motioneye/issues/930

ccrisan, 2018-10-04:
@jasaw I thought @jazeph is referring to streaming/playback which shouldn't be impacted by the hardware encoding. Or?

jazeph, 2018-10-04:
Both thanks for your response.

This is the exact setup:
- Raspbian Stretch lite 2018-06-27
- motionEye 0.39.3
- motion 4.1.1
- ffmpeg 3.2.10-1~deb9u1+rpt2

It seems like recording of the video is fine, because if I download a movie and play it locally, it is ok. But the download speed is extremely slow, which makes direct playback from the recorded movies impossible.

@jasaw So I would think that GPU hardware acceleration has nothing to do with this? And isn't it true that if hardware acceleration isn't enabled, I won't be able to choose the OMX movie formats?

The environment (board, wired network, sd card) is exactly the same, the difference is the OS on the sd card. And on motionEyeOS the download/playback is just fine where the standalone installation on Raspbian has slow download speed.

Thanks for thinking along.

jasaw, 2018-10-05:
I thought the video playback itself was slow, but @jazeph can play it locally, so yes, this is **not** an OMX issue.

@jazeph Just want to clarify, you're playing back videos on a web browser via motionEye web interface?

Did you use the same web browser?
SSH into the device, run `top` command and see which process is hogging your CPU.

jazeph, 2018-10-05:
@jasaw Yes using the same browser for playback. But different browsers gave the same results both for direct playback and download: ok on motionEyeOS, poor on Raspbian. If I wait until movie playback is finished and start playback again, it is ok but this is probably due to browser caching.

`top` doesn't say anything special. `motion` is constantly using about 70-80%, probably because it's constantly retrieving data from my IP cam. But this was also all the same on motionEyeOS.

I even made sure that `motion` was not running and still the download speed was very poor. I tried to `scp` a movie file and this is a lot faster. So I don't think it is a network related issue.

`htop` gives me this:
[image]

Isn't it strange to see this many `meyectl` processes?

Maybe I should look into the webserver. In motionEysOS's code I see that Tornado 4.5.1 is used and the version on my Raspbian installation is 5.1.1. I will try installing version 4.5.1...

jazeph, 2018-10-05:
Unfortunately Tornado 4.5.1 makes no difference... Reverted back to 5.1.1

jazeph, 2018-10-05:
Also just downgraded to motionEye 0.39.2 because that is the version that is in the latest motionEyeOS build, but again with no luck. Reverted back to 0.39.3

orsfey, 2018-10-27:
I have same problem, montioneye on raspberry pi little or desktop with official ffmpeg 3.2.10  motion 4.1.1.1 motion 0.39.3 installed, not work, online download very slow, and even can't playback, but I try it on motioneyeos version it played smoothly, I try install ffmpeg 3.3.5 or later version and disable zero copy from source install, still no work, I don't know how to fix it, I have try versions for almost one week, still can't playback on web no matter mp4,kvm,avi, or else which extension
[image]

jazeph, 2018-10-31:
The issue is still here as well. Also the "Media format unsupported" when trying to play from a mobile device (not from desktop). I think it has something to do with the webserver, because when I stopped motionEye and started another webserver, I noticed that download speed was much higher.

andresanches, 2019-01-29:
I had a similar issue when trying to view recorded footage (stream, not download) on my android phone (on chrome). I noticed that the playback _almost_ started, even showing recording length in minutes and seconds but then crapped out. I also noticed that on an actual computer wired to the ethernet streaming stuttered a lot because buffering was slower than the playback speed (strange for a gigabit network), but ultimately it worked. This indicated it was not a corruption issue but some sort of bottleneck in the code causing a timeout and as it turns out, at least for me, it was

Looking [at the code here](https://github.com/ccrisan/motioneye/blob/28a663e8fa774e5c2f9aa4337affdf3fa879a9b9/motioneye/handlers.py#L1608) I found that the `MoviePlaybackHandler` uses the `/tmp` directory heavily, which in my vanilla debian stretch lite was set up on the good old hard drive. So I changed it to be mounted `tmpfs` instead, as per [this suggestion](https://unix.stackexchange.com/a/220851)

Now not only playback works fine on mobile but the stuttering on the computer is also gone

Hope this helps others who run into this issue

swanny1, 2019-02-28:
I seem to be having a similar issue.
I can’t play back on iOS, on windows 10 it is fine and is distorted badly on Android.  I have the codec set to h.264. If I download a file using the download button it appears to append the file name with a ; so instead of file.mp4 I get file.mp4;

1911mc, 2019-11-11:
> I have same problem, montioneye on raspberry pi little or desktop with official ffmpeg 3.2.10 motion 4.1.1.1 motion 0.39.3 installed, not work, online download very slow, and even can't playback, [...]

Has anybody been able to solve this issue, I have exactly the same problem

andresanches, 2019-11-11:
@1911mc try to mount the /tmp folder as tmpfs. See instructions here: https://unix.stackexchange.com/questions/55773/move-tmp-to-ram/220851#220851

1911mc, 2019-11-12:
Hi, thanks for the response, however /tmp already seems to be mounted to 
tmpfs

> [root@meye-5a5b2192 share]# df
> Filesystem           1K-blocks      Used Available Use% Mounted on
> /dev/root               218143    204239         0 100% /
> devtmpfs               1939964         0   1939964   0% /dev
> tmpfs                  1973756        16   1973740   0% /dev/shm
> tmpfs                  1973756      2368   1971388   0% /tmp
> /dev/mmcblk0p1           30642     11906     18736  39% /boot
> /dev/mmcblk0p3        29514228     51720  27940204   0% /data
> /dev/mmcblk0p3        29514228     51720  27940204   0% /home/ftp/sdcard
> /dev/mmcblk0p3        29514228     51720  27940204   0% /home/ftp/storage
> /dev/sda             3844630544  56426780 3592854516   2% /data/media/sda
> /dev/sda             3844630544  56426780 3592854516   2% 
> /home/ftp/storage/sda
Also when I tried to run the suggested command I got this:
[root@meye-5a5b2192 ~]# cp /usr/share/systemd/tmp.mount /etc/systemd/system/
cp: can't stat '/usr/share/systemd/tmp.mount': No such file or directory

Eugene

jazeph, 2019-11-23:
> @1911mc try to mount the /tmp folder as tmpfs. See instructions here: https://unix.stackexchange.com/questions/55773/move-tmp-to-ram/220851#220851

Also tried this and alternative way to mount /tmp as tmpfs, but the result remains the same: playback on iOS results in the previously mentioned error.

JamBalaya56562, 2026-11-08:
Closing, as this was with motionEye 0.39.3 on Python 2 and Raspbian Stretch, and no difference between the two setups was found. Movies were fine once downloaded, so recording worked, and only serving them was slow. The web server and Python stack have all changed since then. Please open a new issue with a current release if movies still load much slower than other files from the same device.

closed 2026-11-08 by JamBalaya56562