motioneye-project/motioneye #1018: Version `CURL_OPENSSL_3' not found

baitun, 2018-10-05:
After update to Ubunto 18.4.1, motioneye doesn't work.  
`service motioneye status` return following error:
```
● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2018-10-04 10:44:02 +08; 1 day 13h ago
  Process: 781 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited, status=1/FAILURE)
 Main PID: 781 (code=exited, status=1/FAILURE)

окт 04 10:44:02 ubuntu-samsung meyectl[781]:     import handlers
окт 04 10:44:02 ubuntu-samsung meyectl[781]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 30, in <module>
окт 04 10:44:02 ubuntu-samsung meyectl[781]:     import config
окт 04 10:44:02 ubuntu-samsung meyectl[781]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 37, in <module>
окт 04 10:44:02 ubuntu-samsung meyectl[781]:     import uploadservices
окт 04 10:44:02 ubuntu-samsung meyectl[781]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/uploadservices.py", line 26, in <module>
окт 04 10:44:02 ubuntu-samsung meyectl[781]:     import pycurl
окт 04 10:44:02 ubuntu-samsung meyectl[781]: ImportError: /usr/lib/x86_64-linux-gnu/libcurl.so.4: version `CURL_OPENSSL_3' not found (required by /usr/local/l
окт 04 10:44:02 ubuntu-samsung systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
окт 04 10:44:02 ubuntu-samsung systemd[1]: motioneye.service: Failed with result 'exit-code'.
```

baitun, 2018-10-05:
I've found the solution: https://github.com/curl/curl/issues/2433#issuecomment-392661742
Issue can be closed.

closed 2018-10-05 by baitun

akidburn, 2019-05-22:
There is a problem with this solution, because installing libcurl3 uninstalls apache2, and thats bad