motioneye-project/motioneye #1023: performance problems on large installation - Debian 9.5 / 27 Cams / 24 Core

Bartlamweg, 2018-10-09:
Hello!

first of all - great Project! I build some smaller installations with motioneye already and it is extremely useful!
But now to my Problem:

I have installed before a Motioneye Server on Debian 9.1 which was quite large already with its 16 Cams, mixed of Raspberry cams and cheap china cams. So far it was running ok for me, since i am using this to recognize plate numbers i had no big performance issues. But i see on my Server alwas full load of 7 when i check top. This setup is running good since about 1 year. The machine has a 4 core i5 processor.

Now i was thinking i will use Motioneye Server on Debian 9.5 to make a larger installation, where i  need the motion detection and quite good videos of the things that happen.
I decided to use a 2 CPU Proliant Server with 24 Core to beat the big load problem of my prior installation. But it is not as i have hoped: now this server runs with 27 cams, all china with full hd rtsp stream. It is not really performant, loosing frames while recording with maby 0,5 to 1 frame per sec, the load of the Server is at actual unbelievable 52 !, CPU are running permanent with 100%. The total network throughput with all this cams together is 20 to 60 Mbps, while not recording, not really higher when recording. I have a 10Gbit connection and it does not need 100Mbit :)

My Question is if i am doing something so badly wrong? 
Is there such a big difference in Cameras? 
I expected a much bigger bandwith for the camera streams but it is at about 1-2Mbit per cam.

The movie format h.264 i cannot use at all, the performance goes down even more, i use msmpeg4v2, this is the only way i can accept it running at all. I had the same issue already on my 16 cam install.

I would be very thankful for hints how to make this installation better, since it has to get more cams in future, if there is a way.

Thank You.

kleini, 2018-10-09:
The high CPU load mostly comes from the background motion processes and not from motioneye. Therefore optimizations needs to be done on motion.

jasaw, 2018-10-12:
@Bartlamweg Maybe try offloading the decoding of the 27 full HD RTSP streams to the GPU, if you have one. You may have to change the default h264 decoder to a HW accel version in ffmpeg and recompile.

For motion detection, try merging the SSE optimizations back into motion and recompile. See https://github.com/Motion-Project/motion/pull/498

With H264 encoding on movie format, try offloading it to the GPU too.

Bartlamweg, 2019-01-14:
Thank you for your comments, this is going to deep for my knowledge. Due to my inactivity i am closing this issue now. 
Thanks again :)

closed 2019-01-14 by Bartlamweg