motioneye-project/motioneye #1024: Scheduled captures

vstanese, 2018-10-09:
Hello!
Is it possible to have a camera capture stills based on a schedule?
I need to do some monitoring and need to have 20 pictures taken at certain hours (and minutes) throughout the day.
Can motioneye be configured accordingly?
Best regards,
Voicu!

SiliconKite, 2019-07-18:
I believe camera stills can only be captured based on schedule if they are triggered by motion detection.
As far as I can see and tests, the interval snapshots (i.e pics taken at some set interval) can't be set to happen within an schedule
Would it be possible to add this feature?
Thanks

zagrim, 2019-07-19:
Is this duplicate of #883 ?

SiliconKite, 2019-07-21:
I think so. Thanks

starbasessd, 2021-11-23:
The currently 'easiest' way to handle this is to have 2 settings, one with the capture settings 'on' and one with the capture settings 'off', and use a crontab or action button (or both) to do what Working Hours does, en effect. I've posted similar methods here for other issues.

JamBalaya56562, 2026-11-10:
Closing as a duplicate of #883, as agreed above: both ask for Still Images to follow a schedule. Until then, two camera setups switched by cron or action buttons, as starbasessd described, do the same.

closed 2026-11-10 by JamBalaya56562