motioneye-project/motioneye #1025: How to send Image/Movie or Filename via WEBHOOK

tsprivate, 2018-10-15:
I have setup the motion detection and it woks fine.

I want to inform my Home Automation system about the motion and provide the evidence either as snapshot (directly) or via filename (jpg or mp4 file).

How can I configure webhook to transfer the necessary data?

My Home Automation system is already detecting the webhook but there is no content sent (yet)

dgomes, 2018-11-12:
webhook doesn't send any image

henning-kln, 2024-02-28:
Can you describe how do you did it?

JamBalaya56562, 2026-11-07:
Closing as solved, as you confirmed: the webhook does not send the image itself, and `%f` in the URL passes the full path of the file. To get pictures into Telegram, as asked above, motionEye can send them itself with the Telegram notification option under Motion Notifications.

closed 2026-11-07 by JamBalaya56562