motioneye-project/motioneye #1026: Can not record Videos on RTSP Stream...

byEliteHacks, 2018-10-16:
[image]
[image]
[image]
[image]

Some screenshots might help...

Harrie1966, 2018-10-24:
if you have not installed a raspberry pi. Then just try Mp4. That works fine for me.

ccrisan, 2018-10-28:
@byEliteHacks what happens if you click the movies icon instead of the pictures one, on your camera frame?

jimmyeao, 2019-10-21:
Did you find an answer to this? None of my RTSP cameras are recording video on motion, but they will save images. already tried MP4. Motioneye installed on Ubuntu 18.04

clydefrog76, 2020-01-29:
Same issue here. RTSP stream works fine, I can take snapshots to the storage folder but it will not record anything. I tried all the motion settings including auto threshold and nothing is recorded. Any suggestions?

marc0nline, 2020-02-03:
> Same issue here on Ubuntu 18.04.03 LTS. RTSP stream works fine, I can take snapshots to the storage folder but it will not record anything. I tried all the motion settings including auto threshold and nothing is recorded. Any suggestions?

I'm having same issue.

sher-zaman, 2020-02-12:
Just change the movie format to H.264 (.mp4), This solved this issue for me @ Ubuntu 18.04.04 LTS.

oyhel, 2020-02-22:
I had the same problem. Seems the issue is permissions (I assume). I added a folder CAM in my share (/mnt/myshare/CAM/), then chmod 777 and specified  root directory to /CAM in motioneye. Together with changing to H.264 (mp4) movies started to record to the share. Next problem; files are huge...

sher-zaman, 2020-02-24:
I am also facing same issue with huge file size, 1 hr recording file is sizing at 3.6 GB.

closed 2020-03-05 by fynnsh