motioneye-project/motioneye #1027: Ubuntu install instructions need updated.

blindrain, 2018-10-16:
hardware [x86, ARM, other]: x86_x64
operating system ubuntu 18.04
Step 1, and 2 is not able to be completed on ubuntu 18.04 all 3 packages do not exist.

Klausoleum, 2018-10-17:
I did it on the basis of this manual and it works wonderfully (64bit laptop, i7)
only the timelapse does not work yet

[https://github.com/ccrisan/motioneye/wiki/Install-On-Ubuntu](url)

blindrain, 2018-10-17:
I'm setting up a vm to show you what i'm running into.

This is a fresh install of ubuntu. 18.04

[image: iso file used for vm.PNG]

[image: Default repo.PNG]

Apps selected = none
[image: apps selected for install..PNG]

First Boot after fresh install

[image: First boot after install.PNG]

Entering Command Followed by results.

[image: Command after login..PNG]

[image: result from command.PNG]

__
Robert Whitacre
(317)449-9320
The IT Method
Amazon Wish List
<http://www.amazon.com/gp/registry/wishlist/3BKCUEJWNAAC2/ref=cm_wl_rlist_go_o>
Need to Send Me an Encrypted File
<https://senditonthenet.com/secure/drop/robert@theitmethod.com>

blindrain, 2018-10-17:
I had to run this command followed by your intructions for this to work. 

`sudo add-apt-repository "deb http://archive.ubuntu.com/ubuntu $(lsb_release -sc) main universe restricted multiverse"`

closed 2018-10-17 by blindrain