motioneye-project/motioneye #1029: High CPU Usage and Frame Rate in Hundreds Sometimes Thousands

Raptor405, 2018-10-20:
[image]
[image]

Been imaging and re-imaging SD cards to get Ubuntu-Mate and motioneye working with motion 4.0 on my raspberry-pi B. Finally got it all working (sorta) and from the pictures you can see the frame counter is all wack but I have it set at 10 fps. Now it only displays 10 FPS and not a 1000+ but through all my other installs of motioneye it has never done this and never used so much CPU (50% and more). It will bounce from displaying a couple hundred to in the thousands. It causes my PI to run at 70 degrees Celsius with a large after market heat sink(no fan at the moment). 

Somethings wrong and I can't figure it out. I ran higher resolution and frame rate while only using 10% of the CPU before. I am on the latest motioneye, ubuntu, and all. 

Any ideas? I apologize if I missed aa thread about this already, I only saw threads about low FPS, not high.

SealedJoy, 2018-11-14:
Returns: 
ffmpeg version N-92427-gae43235

Raptor405, 2018-11-18:
I get a 3.2-2+rpi1~xenial1.7 version returned back to me.

jasaw, 2018-11-21:
Can you guys check what the CPU load is in these 2 scenarios:
1. Close all connections to the motionEye device, except for SSH connection. That means close your web browser as well, to stop the live streaming. After closing your live stream, it takes a few minutes for CPU load to drop (if it drops at all). Disable constant video recording if necessary. Check the CPU load from the SSH connection. This should at least isolate which part of the software stack is loading the CPU.
2. Identify whether MotionEye or Motion software that is loading the CPU by running Motion **without** MotionEye. Stop MotionEye `systemctl stop motioneye`. Run Motion on its own: `motion -c /etc/motioneye/thread-1.conf -n`.

SealedJoy, 2018-11-21:
Will try and report back asap! :)

JamBalaya56562, 2026-11-09:
The checks jasaw asked for did not come back, and there was no news since 2018, so closing. If the load or the frame counter is still off on a current version, please open a new issue saying whether Motion alone (without motionEye) shows the same load.

closed 2026-11-09 by JamBalaya56562