motioneye-project/motioneye #1030: Motion commands only process when restarting

jstride, 2018-10-20:
I have a curl command I run on event end to send via telegram using curl. The behaviour is really strange that it seems to cache all events and send them when I reapply the config via the front end. I'm running the latest version using the official docker container. I don't see anything in the logs

DaniW42, 2020-08-24:
I am not sure about your problem, but I noticed that my telegram-script runs twice as soon as I save the config. But I think this due to the reinitialize of the camera which leads to an motion event (motioneye thinks there was a motion as the camera picture changes).

JamBalaya56562, 2026-11-10:
Closing, as this got no follow-up and was with the 2018 Docker image. Saving the settings restarts Motion, which can itself fire an event, as DaniW42 noticed; events held back until then were not reported again. Please open a new issue with `motion.log` if commands still run late with a current release.

closed 2026-11-10 by JamBalaya56562