motioneye-project/motioneye #1040: Feature request: Delete locally after successful upload te remote server.

vandalon, 2018-10-28:
Not so much an issue as a feature request.

It would be nice to have an option to delete local recording after successful upload to remote server.

adamtbeames, 2018-11-14:
I'm after this feature too.
I'd love to be able to also tier storage if possible.
Tier 1 = local (e.g. 1 day)
Tier 2 = AWS S3 (e.g. 1 week)
Tier 3 = AWS Glacier (e.g. 1 month)

This relates to:
https://github.com/ccrisan/motioneye/pull/1063
https://github.com/ccrisan/motioneye/issues/1066

JamBalaya56562, 2026-11-09:
Done in #3374 (0.45.0b1): the "Remove Files After Upload" option under the camera's file storage settings deletes a movie or picture locally once the upload service has uploaded it successfully. Tiered storage (S3, Glacier) would be a separate request. Closing.

closed 2026-11-09 by JamBalaya56562