motioneye-project/motioneye #1041: Wiki: Installation Instruction Updates for Linux Mint 19

BrewBuilding, 2018-10-28:
Informational update for installing MotionEye on the Linux Mint 19 with systemd:

Install Linux Mint 19 with default settings.  Get all system updates.

Follow the [Installation instructions](https://github.com/ccrisan/motioneye/wiki/Installation) on the MotionEye GitHub wiki.

Then, follow the [Ubuntu instructions](https://github.com/ccrisan/motioneye/wiki/Install-On-Ubuntu) on the MotionEye GitHub wiki.  The following notes show deviations from and additions to the Ubuntu instructions.

Mint doesn't install *pip* by default, and it  includes both Python2 and Python3 by default, so, install and update *pip* for both versions of Python.  Note that because both *pip2* and *pip3* are active, and MotionEye wants Python2, you need to use *pip2* where the instructions say *pip*.

  * `sudo apt install python-pip`
  * `sudo apt install python3-pip`
  * `sudo pip2 install --upgrade pip`
  * `sudo pip3 install --upgrade pip`

Mint also doesn't install *setuptools* by default, so, install *setuptools* for both versions of Python.

  * `sudo -H pip2 install setuptools`
  * `sudo -H pip3 install setuptools`

Under Mint, some of the prerequisite packages have to come through the package installer rather than *pip*.  So install the rest of the prerequisite packages like this.

  * `sudo apt install curl libcurl4 python-pycurl python3-pycurl`
  * `sudo apt install v4l-utils ffmpeg motion`
  * `sudo apt install python-pillow python3-pillow` version 5.1.0-1
  * `sudo apt install python-jinja2 python3-jinja2` version 2.10-1
  * `sudo apt install python-tornado python3-tornado` version 4.5.3-1
  * `sudo apt install python-dev python3-dev`

The rest of the Ubuntu instructions should work as printed, if you substitute *pip2* for *pip*.

  * `sudo -H pip2 install motioneye`
  * `sudo mkdir -p /etc/motioneye`
  * `sudo cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf`
  * `sudo mkdir -p /var/lib/motioneye`
  * `sudo cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service`
  * `sudo systemctl daemon-reload`
  * `sudo systemctl enable motioneye`
  * `sudo systemctl start motioneye`

nicaroxo, 2020-01-20:
Life saver! On my system BPI-M1 buster 10 was getting some errors following the old steps on debian... These steps now should be recomended for Debian Buster also, mabe other new debs too. Thanks for your awesome work! Congrats!

nicaroxo, 2020-05-25:
Update... Also works for Ubuntu 20.04LTS!

closed 2022-03-19 by MichaIng