motioneye-project/motioneye #1042: Web refuses connection

Ted-Corrrell, 2018-10-29:
I had installed MotioneyeOS on a raspberry Pi2, and installed Motioneye on a debian server. I had both running, and I tried adding the pi as a network camera, but something with the pi stopped working. I had to reinstall MotioneyeOS on the Pi, and once I had it working again I went to log into motioneye on the server, and the web page refused the connection. I restarted, hoping that would fix things; when it didn't I reinstalled Motioneye. I have tried to open it from different computers, and they all get refused. I tried looking at the logs, but there isn't any log file for Motioneye, even after setting the log level through the config file to 'debug'. Using systemctl status motioneye and journalctl - motioneye I was able to see some status messages, but there were never errors or messages about refusing a connection. I don't know what else to try. Any help is much appreciated. If there's any additional info I can give let me know. 
Thanks

jasaw, 2018-10-29:
Your MotionEye server is configured to listen on port 8765. Did you try reaching it on that port?

Ted-Corrrell, 2018-10-29:
Well, I feel dumb now. Thank you! I don't remember having to specify the port for Motioneye on the server before I had to reinstall MotioneyeOS on the pi, but I might have.

closed 2018-10-29 by Ted-Corrrell