motioneye-project/motioneye #1044: MotionEye keeps restarting

xu4user, 2018-11-01:
Here is my setup:

XU4 with SSD for external storage plugged in via the USB3 port.
Running off of eMMC drive
Ubuntu 16.04
Latest version of Motioneye

I have 2 IP cameras. I have them setup on MotionEye and they work. However, they are constantly getting the X thru the camera showing up. When I look at the syslog, I see the following error:

 09:59:13 odroid rsyslogd-2007: action 'action 10' suspended, next retry is Thu Nov  1 10:00:43 2018 [v8.16.0 try http://www.rsyslog.com/e/2007 ]
Nov  1 09:59:13 odroid meyectl[641]:  WARNING: Connect error on fd 21: ECONNREFUSED
Nov  1 09:59:23 odroid meyectl[641]: message repeated 82 times: [  WARNING: Connect error on fd 21: ECONNREFUSED]
Nov  1 09:59:23 odroid meyectl[641]:    ERROR: motion not running, starting it
Nov  1 09:59:23 odroid motion: [501056] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Nov  1 09:59:23 odroid motion: [501056] [ALR] [ALL] conf_cmdparse: Unknown config option "webcontrol_parms"
Nov  1 09:59:23 odroid motion: [501056] [NTC] [ALL] config_thread: Processing config file thread-1.conf
Nov  1 09:59:23 odroid motion: [501056] [NTC] [ALL] config_thread: Processing config file thread-2.conf
Nov  1 09:59:23 odroid motion: [501056] [NTC] [ALL] motion_startup: Motion 3.2.12+git20140228 Started with SDL support
Nov  1 09:59:23 odroid motion: [501056] [NTC] [ALL] motion_startup: Logging to syslog
Nov  1 09:59:23 odroid motion: [501056] [NTC] [ALL] motion_startup: Using default log type (ALL)
Nov  1 09:59:23 odroid motion: [501056] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Nov  1 09:59:23 odroid motion: [1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.254.64:554/11)
Nov  1 09:59:23 odroid motion: [2] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.254.82:554/11)
Nov  1 09:59:25 odroid meyectl[641]:  WARNING: Connect error on fd 22: ECONNREFUSED
Nov  1 09:59:25 odroid meyectl[641]:  WARNING: Connect error on fd 21: ECONNREFUSED
Nov  1 09:59:25 odroid meyectl[641]: message repeated 2 times: [  WARNING: Connect error on fd 21: ECONNREFUSED]
Nov  1 09:59:25 odroid motion: [2] [WRN] [NET] netcam_read_rtsp_image: unusual frame size of 3110400!: Operation now in progress
Nov  1 09:59:25 odroid motion: [2] [ALR] [NET] netcam_handler_loop: Camera handler thread [3] started
Nov  1 09:59:25 odroid meyectl[641]:  WARNING: Connect error on fd 21: ECONNREFUSED
Nov  1 09:59:25 odroid meyectl[641]: message repeated 2 times: [  WARNING: Connect error on fd 21: ECONNREFUSED]
Nov  1 09:59:25 odroid meyectl[641]:  WARNING: Connect error on fd 22: ECONNREFUSED
Nov  1 09:59:26 odroid meyectl[641]:  WARNING: Connect error on fd 22: ECONNREFUSED
Nov  1 09:59:26 odroid motion: [1] [ALR] [NET] netcam_handler_loop: Camera handler thread [4] started
Nov  1 09:59:26 odroid meyectl[641]:  WARNING: Connect error on fd 22: ECONNREFUSED
Nov  1 09:59:36 odroid meyectl[641]:    ERROR: connection problem detected for mjpg client for camera 1 on port 8081
Nov  1 09:59:36 odroid meyectl[641]:  WARNING: Connect error on fd 21: ECONNREFUSED
Nov  1 09:59:45 odroid meyectl[641]: message repeated 79 times: [  WARNING: Connect error on fd 21: ECONNREFUSED]
Nov  1 09:59:45 odroid meyectl[641]:    ERROR: motion not running, starting it
Nov  1 09:59:45 odroid motion: [501056] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Nov  1 09:59:45 odroid motion: [501056] [ALR] [ALL] conf_cmdparse: Unknown config option "webcontrol_parms"
Nov  1 09:59:45 odroid motion: [501056] [NTC] [ALL] config_thread: Processing config file thread-1.conf
Nov  1 09:59:45 odroid motion: [501056] [NTC] [ALL] config_thread: Processing config file thread-2.conf
Nov  1 09:59:45 odroid motion: [501056] [NTC] [ALL] motion_startup: Motion 3.2.12+git20140228 Started with SDL support
Nov  1 09:59:45 odroid motion: [501056] [NTC] [ALL] motion_startup: Logging to syslog
Nov  1 09:59:45 odroid motion: [501056] [NTC] [ALL] motion_startup: Using default log type (ALL)
Nov  1 09:59:45 odroid motion: [501056] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Nov  1 09:59:45 odroid motion: [2] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.254.82:554/11)
Nov  1 09:59:45 odroid motion: [1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.254.64:554/11)
Nov  1 09:59:47 odroid meyectl[641]:  WARNING: Connect error on fd 22: ECONNREFUSED
Nov  1 09:59:47 odroid meyectl[641]:  WARNING: Connect error on fd 21: ECONNREFUSED
Nov  1 09:59:47 odroid meyectl[641]: message repeated 2 times: [  WARNING: Connect error on fd 21: ECONNREFUSED]
Nov  1 09:59:47 odroid motion: [2] [WRN] [NET] netcam_read_rtsp_image: unusual frame size of 3110400!: Operation now in progress
Nov  1 09:59:47 odroid motion: [2] [ALR] [NET] netcam_handler_loop: Camera handler thread [3] started
Nov  1 09:59:47 odroid meyectl[641]:  WARNING: Connect error on fd 22: ECONNREFUSED
Nov  1 09:59:48 odroid meyectl[641]: message repeated 3 times: [  WARNING: Connect error on fd 22: ECONNREFUSED]
Nov  1 09:59:48 odroid motion: [1] [ALR] [NET] netcam_handler_loop: Camera handler thread [4] started
Nov  1 10:12:44 odroid rsyslogd-2007: action 'action 10' suspended, next retry is Thu Nov  1 10:14:14 2018 [v8.16.0 try http://www.rsyslog.com/e/2007 ]

xu4user, 2018-11-01:
Ok, I fixed that error, but now I have this in the syslog:

Nov  1 10:57:15 odroid motion: [501056] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Nov  1 10:57:15 odroid motion: [501056] [ALR] [ALL] conf_cmdparse: Unknown config option "webcontrol_parms"
Nov  1 10:57:15 odroid motion: [501056] [NTC] [ALL] config_thread: Processing config file thread-1.conf
Nov  1 10:57:15 odroid motion: [501056] [NTC] [ALL] config_thread: Processing config file thread-2.conf
Nov  1 10:57:15 odroid motion: [501056] [NTC] [ALL] motion_startup: Motion 3.2.12+git20140228 Started with SDL support
Nov  1 10:57:15 odroid motion: [501056] [NTC] [ALL] motion_startup: Logging to syslog
Nov  1 10:57:15 odroid motion: [501056] [NTC] [ALL] motion_startup: Using default log type (ALL)
Nov  1 10:57:15 odroid motion: [501056] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Nov  1 10:57:15 odroid motion: [1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.254.64:554/11)
Nov  1 10:57:15 odroid motion: [2] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.254.82:554/11)
Nov  1 10:57:17 odroid motion: [2] [WRN] [NET] netcam_read_rtsp_image: unusual frame size of 3110400!: Operation now in progress
Nov  1 10:57:17 odroid motion: [2] [ALR] [NET] netcam_handler_loop: Camera handler thread [3] started
Nov  1 10:57:17 odroid meyectl[3921]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov  1 10:57:17 odroid ntpd[1153]: 91.189.89.198 local addr 192.168.254.75 -> <null>
Nov  1 10:57:17 odroid motion: [1] [ALR] [NET] netcam_handler_loop: Camera handler thread [4] started
Nov  1 10:57:21 odroid ntpd[1153]: 91.189.89.199 local addr 192.168.254.75 -> <null>
Nov  1 10:57:31 odroid meyectl[3921]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov  1 10:57:36 odroid meyectl[3921]: message repeated 25 times: [  WARNING: Connect error on fd 15: ECONNREFUSED]
Nov  1 10:57:37 odroid meyectl[3921]:    ERROR: motion not running, starting it
Nov  1 10:57:37 odroid motion: [501056] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Nov  1 10:57:37 odroid motion: [501056] [ALR] [ALL] conf_cmdparse: Unknown config option "webcontrol_parms"
Nov  1 10:57:37 odroid motion: [501056] [NTC] [ALL] config_thread: Processing config file thread-1.conf
Nov  1 10:57:37 odroid motion: [501056] [NTC] [ALL] config_thread: Processing config file thread-2.conf
Nov  1 10:57:37 odroid motion: [501056] [NTC] [ALL] motion_startup: Motion 3.2.12+git20140228 Started with SDL support
Nov  1 10:57:37 odroid motion: [501056] [NTC] [ALL] motion_startup: Logging to syslog
Nov  1 10:57:37 odroid motion: [501056] [NTC] [ALL] motion_startup: Using default log type (ALL)
Nov  1 10:57:37 odroid motion: [501056] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Nov  1 10:57:37 odroid motion: [2] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.254.82:554/11)
Nov  1 10:57:37 odroid motion: [1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.254.64:554/11)
Nov  1 10:57:39 odroid motion: [2] [WRN] [NET] netcam_read_rtsp_image: unusual frame size of 3110400!: Operation now in progress
Nov  1 10:57:39 odroid motion: [2] [ALR] [NET] netcam_handler_loop: Camera handler thread [3] started
Nov  1 10:57:40 odroid motion: [1] [ALR] [NET] netcam_handler_loop: Camera handler thread [4] started
Nov  1 10:58:12 odroid meyectl[3921]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov  1 10:58:12 odroid rsyslogd-2007: action 'action 10' suspended, next retry is Thu Nov  1 10:59:12 2018 [v8.16.0 try http://www.rsyslog.com/e/2007 ]
Nov  1 10:58:17 odroid meyectl[3921]: message repeated 25 times: [  WARNING: Connect error on fd 15: ECONNREFUSED]
Nov  1 10:58:19 odroid meyectl[3921]:    ERROR: motion not running, starting it
Nov  1 10:58:19 odroid motion: [501056] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Nov  1 10:58:19 odroid motion: [501056] [ALR] [ALL] conf_cmdparse: Unknown config option "webcontrol_parms"
Nov  1 10:58:19 odroid motion: [501056] [NTC] [ALL] config_thread: Processing config file thread-1.conf
Nov  1 10:58:19 odroid motion: [501056] [NTC] [ALL] config_thread: Processing config file thread-2.conf
Nov  1 10:58:19 odroid motion: [501056] [NTC] [ALL] motion_startup: Motion 3.2.12+git20140228 Started with SDL support
Nov  1 10:58:19 odroid motion: [501056] [NTC] [ALL] motion_startup: Logging to syslog
Nov  1 10:58:19 odroid motion: [501056] [NTC] [ALL] motion_startup: Using default log type (ALL)
Nov  1 10:58:19 odroid motion: [501056] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Nov  1 10:58:19 odroid motion: [1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.254.64:554/11)
Nov  1 10:58:19 odroid motion: [2] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.254.82:554/11)
Nov  1 10:58:21 odroid motion: [2] [WRN] [NET] netcam_read_rtsp_image: unusual frame size of 3110400!: Operation now in progress
Nov  1 10:58:21 odroid motion: [2] [ALR] [NET] netcam_handler_loop: Camera handler thread [3] started
Nov  1 10:58:22 odroid motion: [1] [ALR] [NET] netcam_handler_loop: Camera handler thread [4] started

jasaw, 2018-11-03:
Looks like MotionEye was running as expected, did not restart at all. It was `motion` software that was restarting. Follow [this guide](https://github.com/ccrisan/motioneye/wiki/Report-Motion-Issue) to run motion at debug log level to help figure out what's causing motion to crash.

xu4user, 2018-11-05:
Ok, here are the files.

Please let me know what to do because it just doesn't work correctly at all.

[thread-1.txt](https://github.com/ccrisan/motioneye/files/2550375/thread-1.txt)
[thread-2.txt](https://github.com/ccrisan/motioneye/files/2550376/thread-2.txt)
[motioneye.txt](https://github.com/ccrisan/motioneye/files/2550380/motioneye.txt)
[motion.log](https://github.com/ccrisan/motioneye/files/2550381/motion.log)

ccrisan, 2018-11-05:
@xu4user I strongly recommend upgrading to Motion 4.x. You're using an ancient Motion version:

    2018-11-05 15:18:35: [sendmail]    DEBUG: using motion version 3.2.12+git20140228

xu4user, 2018-11-08:
How do I update motion? I did download motion_4.1.1-1.1build1_armhf.deb and tried to install it, but it comes back and says:
Error: Dependency is not satisfiable: libavcodec58 (>=7:4.0) | libavcodec-extra58 (>=7:4.0)

xu4user, 2018-11-08:
Then I tried this: 

sudo dpkg -i motion_4.1.1-1.1build1_armhf.deb
(Reading database ... 219233 files and directories currently installed.)
Preparing to unpack motion_4.1.1-1.1build1_armhf.deb ...
Unpacking motion (4.1.1-1.1build1) over (3.2.12+git20140228-8build1) ...
dpkg: dependency problems prevent configuration of motion:
 motion depends on libavcodec58 (>= 7:4.0) | libavcodec-extra58 (>= 7:4.0); however:
  Package libavcodec58 is not installed.
  Package libavcodec-extra58 is not installed.
 motion depends on libavdevice58 (>= 7:4.0); however:
  Package libavdevice58 is not installed.
 motion depends on libavformat58 (>= 7:4.0); however:
  Package libavformat58 is not installed.
 motion depends on libavutil56 (>= 7:4.0); however:
  Package libavutil56 is not installed.
 motion depends on libswscale5 (>= 7:4.0); however:
  Package libswscale5 is not installed.

So then I tried this:
sudo apt-get install libavcodec58 libavcodec-extra58
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package libavcodec58 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package libavcodec-extra58 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'libavcodec58' has no installation candidate
E: Package 'libavcodec-extra58' has no installation candidate

Not sure what to do at this point.

xu4user, 2018-11-08:
This is what is showing in the motion.log file:

odroid@odroid:~$ more motion.log
[501056] [ALR] [ALL] conf_load: Configfile motion.log not found - trying defaults.: No such file or directory
[501056] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[501056] [NTC] [ALL] config_thread: Processing config file thread-1.conf
[501056] [NTC] [ALL] config_thread: Processing config file thread-2.conf
[501056] [ALR] [ALL] conf_cmdparse: Unknown config option "webcontrol_parms"
[501056] [NTC] [ALL] motion_startup: Motion 3.2.12+git20140228 Started with SDL support
[501056] [NTC] [ALL] motion_startup: Logging to syslog
[501056] [NTC] [ALL] motion_startup: Using default log type (ALL)
[501056] [NTC] [ALL] motion_startup: Using log type (ALL) log level (NTC)
[501056] [NTC] [ENC] ffmpeg_init: ffmpeg LIBAVCODEC_BUILD 3685476 LIBAVFORMAT_BUILD 3680357
[0] [NTC] [ALL] main: Thread 1 is from thread-1.conf
[0] [NTC] [ALL] main: Thread 1 is device: rtsp://192.168.254.64:554/11 input -1
[0] [NTC] [ALL] main: Stream port 8081
[0] [NTC] [ALL] main: Thread 2 is from thread-2.conf
[0] [NTC] [ALL] main: Thread 2 is device: rtsp://192.168.254.82:554/11 input -1
[0] [NTC] [ALL] main: Stream port 8085
[1] [NTC] [ALL] motion_init: Thread 1 started , motion detection Enabled
[1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.254.64:554/11)
[0] [NTC] [ALL] main: Waiting for threads to finish, pid: 2477
[2] [NTC] [ALL] motion_init: Thread 2 started , motion detection Enabled
[2] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.254.82:554/11)
[0] [NTC] [STR] httpd_run: motion-httpd testing : IPV4 addr: 127.0.0.1 port: 7999
[0] [NTC] [STR] httpd_run: motion-httpd Bound : IPV4 addr: 127.0.0.1 port: 7999
[0] [NTC] [STR] httpd_run: motion-httpd/3.2.12+git20140228 running, accepting connections
[0] [NTC] [STR] httpd_run: motion-httpd: waiting for data on 127.0.0.1 port TCP 7999
[2] [WRN] [NET] netcam_read_rtsp_image: unusual frame size of 3110400!: Operation now in progress
[2] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 1 items
[2] [ALR] [NET] netcam_handler_loop: Camera handler thread [3] started
[2] [ERR] [ALL] get_pgm: Wrong image size 1280x1024 should be 1920x1080: Operation now in progress
[2] [ERR] [ALL] motion_init: Failed to read mask image. Mask feature disabled.
[2] [NTC] [STR] http_bindsock: motion-stream testing : IPV4 addr: 0.0.0.0 port: 8085
[2] [NTC] [STR] http_bindsock: motion-stream Bound : IPV4 addr: 0.0.0.0 port: 8085
[2] [NTC] [ALL] motion_init: Started motion-stream server in port 8085 auth Disabled
[2] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 25 items
[1] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 1 items
[1] [ALR] [NET] netcam_handler_loop: Camera handler thread [4] started
[1] [NTC] [STR] http_bindsock: motion-stream testing : IPV4 addr: 0.0.0.0 port: 8081
[1] [NTC] [STR] http_bindsock: motion-stream Bound : IPV4 addr: 0.0.0.0 port: 8081
[1] [NTC] [ALL] motion_init: Started motion-stream server in port 8081 auth Disabled
[1] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 30 items
[1] [NTC] [EVT] event_new_video FPS 3
[1] [NTC] [ALL] motion_detected: Motion detected - starting event 1
[1] [NTC] [EVT] event_newfile: File of type 1 saved to: /mnt/SSD/Garage/2018-11-08/09-25-58.jpg
[1] [NTC] [EVT] event_newfile: File of type 1 saved to: /mnt/SSD/Garage/2018-11-08/09-25-58.jpg
curl: (7) Failed to connect to 127.0.0.1 port 8765: Connection refused
curl: (7) Failed to connect to 127.0.0.1 port 8765: Connection refused
curl: (7) Failed to connect to 127.0.0.1 port 8765: Connection refused
[1] [NTC] [EVT] event_newfile: File of type 1 saved to: /mnt/SSD/Garage/2018-11-08/09-25-59.jpg
curl: (7) Failed to connect to 127.0.0.1 port 8765: Connection refused
[1] [NTC] [EVT] event_newfile: File of type 1 saved to: /mnt/SSD/Garage/2018-11-08/09-25-59.jpg
curl: (7) Failed to connect to 127.0.0.1 port 8765: Connection refused
[1] [NTC] [EVT] event_newfile: File of type 1 saved to: /mnt/SSD/Garage/2018-11-08/09-25-59.jpg
[1] [NTC] [EVT] event_newfile: File of type 1 saved to: /mnt/SSD/Garage/2018-11-08/09-26-00.jpg
curl: (7) Failed to connect to 127.0.0.1 port 8765: Connection refused
curl: (7) Failed to connect to 127.0.0.1 port 8765: Connection refused
2018-11-08 09:26:14: [sendmail]     INFO: sending email
2018-11-08 09:26:21: [sendmail]     INFO: email sent

ccrisan, 2018-11-09:
@xu4user I believe ubuntu 16.04 comes with those old libs as well as with an old motion. You'll probably need to update your Ubuntu to something more 2018ish.

xu4user, 2018-11-09:
Well there is the v18.04. However, motioneye doesn't even work at all. It won't install properly. So that is why I am using this version.

ccrisan, 2018-11-09:
@xu4user let's fix that then. What's the step where it fails? What guide did you use to install motionEye on Ubuntu 18.04?

xu4user, 2018-11-09:
The INSTALL ON UBUNTU guide from the github website. I have an XU4 that I am installing it on. I could not find any guide to install on that hardware, so I used the ubuntu guide. The distro I am using is Ubuntu MATE 18.04 LTS (v4.0), release date of 5/2018. I did do all the updates and patches before I installed motionEye. One of the features of that distro is FFMPEG with hardware accelerated H.264 decoder. However, one of the bugs reported with that distro is VLC doesn't start due to a video driver compatibility issue. So before I even tried motionEye, I tried the rtsp://ip:554/11 feed using VLC. It only brought up a green screen. Never showed the picture. I tried to install motionEye anyway and after you do the apt-get install motion ffmpeg v4l-utils, then do the apt-get install python and all the libraries, it fails when you try to do the pip install motionEye. It said something was wrong with pillow. I don't remember the exact error. So I tried pip2 install motionEye and that didn't make a difference either. So at that point I just went back to the older distro to try and get things working. I am in no way a ubuntu/Linux guy (although I am learning as I go) and I don't know enough about what is wrong to fix it. I have posted questions on the XU4 forums, but not much help. I would prefer to get it working on the new distro, but again, I am just not experienced enough to know if I am making a mistake or if it is just something buggy in the distro that needs to be fixed or re-compiled or?????

I can re-install the 18.04 distro on my flash card and boot that way and let you know what the error is and post that information. But if VLC doesn't even work, then I don't know how motionEye would.

I am not even sure where to look in what log as to why VLC doesn't work. syslog maybe??

Any help you can offer so that I can get this going would be greatly appreciated.

ccrisan, 2018-11-09:
@xu4user 16.04 is definitely not a solution. If `pillow` is the only package you could not install via `pip`, you can try installing it via `apt-get` as well. It should be named something like `python2-pillow`. Then, you could try `pip install motioneye` and see if it succeeds.

xu4user, 2018-11-09:
Ok, I will try that again and let you know the outcome. Thanks for helping.

xu4user, 2018-11-12:
Ok, I have motionEye installed on 18.04 of ubuntu. Did have some problems with pillow installing but got around that. 

However, the screen comes up GREEN when viewing the video. If I open a browser from a Windows box, it works fine so I know it is not the camera. 

In the motion.log, it says:
[1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rstp://192.168.254.99:554/11)
[1:nc1] [ERR] [EMC] ffmpeg_avodec_log: bad src image pointers
[1:nc1] [ERR] [EMC] ffmpeg_avodec_log: bad src image pointers
[1:nc1] [ERR] [EMC] ffmpeg_avodec_log: bad src image pointers
[1:nc1] [ERR] [EMC] ffmpeg_avodec_log: bad src image pointers
[1:nc1] [ERR] [EMC] ffmpeg_avodec_log: bad src image pointers

And that continues that way to the end of the log.

xu4user, 2018-11-12:
I did try VLC and it just shows a blank screen. If I try VLC from a Windows box, that works fine. 

So it has something to do with the video driver. Not sure how to resolve it. 

Can anyone help?

jasaw, 2018-11-13:
@xu4user I think the bad src image pointers error messages are from ffmpeg library. Unfortunately, we don't know what's happening at that level. Motion developers might be able to give you some pointers. Follow [this guide](https://github.com/ccrisan/motioneye/wiki/Report-Motion-Issue) to report a Motion issue.

xu4user, 2018-11-14:
Ok, I updated to motion 4.2. Now I get this in the log files:

[camera1.txt](https://github.com/ccrisan/motioneye/files/2581069/camera1.txt)
[motion.log](https://github.com/ccrisan/motioneye/files/2581070/motion.log)
[motionconf.txt](https://github.com/ccrisan/motioneye/files/2581071/motionconf.txt)
[syslog.txt](https://github.com/ccrisan/motioneye/files/2581072/syslog.txt)

xu4user, 2018-11-14:
Ok, so now the MOTION team tells me to uninstall motioneye and only go with motion. I am very frustrated with this. I have tried everything that I can think of and have spent many many hours to try and get it to work. The only place I can get it working somewhat is on Ubuntu v16.04.  But that is where I get this error:

curl: (7) Failed to connect to 127.0.0.1 port 8765: Connection refused
curl: (7) Failed to connect to 127.0.0.1 port 8765: Connection refused
curl: (7) Failed to connect to 127.0.0.1 port 8765: Connection refused

So I have no clue what to do. It is extremely frustrating.

jasaw, 2018-11-14:
With motionEye uninstalled, try this:
1. switch to root user.
2. run motion on its own from command line `motion -c thread-1.conf -d 6 -n &> ~/motion.log`. Replace the thread-1.conf file with whichever config file name you used.
3. Open a web browser and point it to the live stream at port 8081.
4. Kill motion and attach the motion.log file here.

Report back what you see.

xu4user, 2018-11-15:
What is did to uninstall motioneye was: 
sudo pip2 uninstall motioneye

I then did a su - (to logon as root)
I ran this: motion -c camera1.conf -d 6 -n &> ~/motion.log
Tried to go to http://camera-ip:8081 it says:
THIS SITE CAN'T BE REACHED

Then I did a CNTRL-C to kill motion.
Here is the motion.log file as an attachment and also posted here. 
root@odroid:~# more motion.log
[0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file camera1.conf
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Motion 4.2 Started
[0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (NTC)
[0:motion] [NTC] [ENC] ffmpeg_global_init: ffmpeg libavcodec version 57.107.100 libavformat version 57.83.100
[0:motion] [NTC] [ALL] translate_init: Language: English
[0:motion] [NTC] [ALL] motion_start_thread: Camera ID: 101 is from camera1.conf
[0:motion] [NTC] [ALL] motion_start_thread: Camera ID: 101 Camera Name: MyCam1 Service: rtsp:
[0:motion] [NTC] [ALL] motion_start_thread: Stream port 0
[0:motion] [NTC] [ALL] main: Waiting for threads to finish, pid: 20141
[1:ml1:MyCam1] [NTC] [ALL] motion_init: Camera 101 started: motion detection Enabled
[1:ml1:MyCam1] [NTC] [VID] vid_start: Opening Netcam RTSP
[1:ml1:MyCam1] [NTC] [NET] netcam_rtsp_ntc: 
[1:ml1:MyCam1] [NTC] [NET] netcam_rtsp_ntc: ******************************************************
[1:ml1:MyCam1] [NTC] [NET] netcam_rtsp_ntc: format than YUV420P.  The image sent is being 
[1:ml1:MyCam1] [NTC] [NET] netcam_rtsp_ntc: trancoded to YUV420P.  If possible change netcam 
[1:ml1:MyCam1] [NTC] [NET] netcam_rtsp_ntc: picture format to YUV420P to possibly lower CPU usage.
[1:ml1:MyCam1] [NTC] [NET] netcam_rtsp_ntc: ******************************************************
[1:ml1:MyCam1] [NTC] [NET] netcam_rtsp_ntc: 
[1:ml1:MyCam1] [NTC] [NET] netcam_rtsp_connect: Normal resolution: Camera (MyCam1) connected
[1:ml1:MyCam1] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 1 items
[2:nc2:MyCam1] [NTC] [NET] netcam_rtsp_handler: Normal resolution: Camera handler thread [2] started
[2:nc2:MyCam1] [NTC] [NET] netcam_rtsp_connect: Normal resolution: Camera (MyCam1) connected
[1:ml1:MyCam1] [NTC] [ALL] mlp_actions: End of event 1
[1:ml1:MyCam1] [NTC] [ALL] motion_loop: Thread exiting
[1:ml1:MyCam1] [NTC] [NET] netcam_rtsp_cleanup: Normal resolution: Shut down complete.
[0:motion] [NTC] [ALL] main: Threads finished
[0:motion] [NTC] [ALL] main: Motion terminating

[motion.log](https://github.com/ccrisan/motioneye/files/2585472/motion.log)

xu4user, 2018-11-15:
Here is also the syslog:
[syslog.txt](https://github.com/ccrisan/motioneye/files/2585533/syslog.txt)

xu4user, 2018-11-15:
Also, I did this command:

ffmpeg -i rtsp://camera-ip:554/11 -r 15 video.avi and that works. I am able to see the video.

xu4user, 2018-11-15:
Ok, so I noticed motion was still using the motioneye.conf, so I got the new motion.conf file and re-ran this command.

motion -c camera1.conf -d 7 -n &> ~/motion.log
Here is the log:

[motion.log](https://github.com/ccrisan/motioneye/files/2585834/motion.log)

closed 2022-03-19 by MichaIng