motioneye-project/motioneye #1045: Strange HTML output accessing my default cam page

clawwd, 2018-11-01:
Hi,

This has got to be something easy. I have never had issues with logging in and seeing the default panel and cameras but now going to the main page has a page that comes up looking like the below output. Tried deleting history and all. Help?

s		?
Fit Frames Vertically		?
Layout Rows		?
Frame Rate Dimmer		?
Resolution Dimmer		?
? General Settings

clawwd, 2018-11-08:
Any ideas guys?

JamBalaya56562, 2026-11-09:
Sorry this got no answer. This looks like the page loaded without its stylesheet, and as the same phone model showed it fine, the cause was likely on that phone (cache, a content blocker or a proxy). Closing, as there was no news since 2018.