motioneye-project/motioneye #1046: No camera on dropdown

olbasoil, 2018-11-02:
I came across the same issue reported (now closed)
https://github.com/ccrisan/motioneye/issues/727

I also get no camera in the dropdown. My cameras is an old Ubiquiti Aircam that works happily in VLC with the address of  rtsp://192.168.1.5: 554/live/ch00_0 

I managed to fix it by entering just rtsp://192.168.1.5 in the ip box. That populated the Camera dropdown (but gave no image). Then I could add the rest of the url which gave the feed.

ccrisan, 2018-11-05:
So what is the problem here?

daniele777, 2019-09-28:
I have same issue After upgrading of motioneyes os i think i'll try to connect on TCP at rtsp://192.168.1.10/live/ch00_0 motioneyes recognized camera but not see nothing ....help

daniele777, 2019-09-28:
Possibile motion Eye bugged i'll try fresh install of

ccrisan, 2019-09-29:
@daniele777 can we see a screenshot of the problem?

daniele777, 2019-09-29:
Cris i can't open cam ,i can  see with app v380 but with motioneyes no ...

daniele777, 2019-09-29:
In future can implement motioneyes with Google home Assistant ?

daniele777, 2019-09-29:
Ok Solve i can't see cam in Google Chrome lol okkk

JamBalaya56562, 2026-11-07:
Thanks for recording the workaround: entering only `rtsp://<address>` first fills the camera list, and adding the rest of the URL then gives the picture. Closing, as nothing is left to do here; please open a new issue if adding an RTSP camera still fails with a current release.

closed 2026-11-07 by JamBalaya56562