motioneye-project/motioneye #1047: How to trigger a detect?

olbasoil, 2018-11-03:
Sorry for a couple of what must be very obvious questions:

I have Motioneye running as an addon in Hassio. I have added a camera.
1) Is there anything in the interface that shows when a camera is recording? I sometimes see a red border round the camera image, but am not sure whether that means it is recording?
2) Can I fire a detection, and so fire a recording? I have got the basic api web interface and can see 'action|makemovie' and 'detection|start' but neither of these seems to work.

My ultimate aim is to 
1 get a motion detection to fire a webhook in Home Assistant

2 Get a door sensor in Home Assistant to begin a recording in Motioneye

and being able to simulate a detection would simplify that.

jasaw, 2018-11-03:
1. On MotionEye web interface, under "Motion Notifications", there's a "Call A Web Hook" option. That should do what you want. You may have to enable "Advanced Settings" for this option to appear.

2. To force a recording, you need to get your Home Assistant to set `emulate_motion` to 1, then set it back to 0 when you want to stop recording. See https://github.com/ccrisan/motioneye/issues/31#issuecomment-370600312

MichaIng, 2026-10-30:
Action buttons for toggling motion detection and a motion event manually have been added with #3468.

closed 2026-10-30 by MichaIng