motioneye-project/motioneye #1048: Doc bug? Installation issue "No module named 'settings'

jgwinner, 2018-11-03:
I have a Raspbian setup; I have some OpenCV dev installations but is a (I think) fairly vanilla Raspian setup. 

When I run motioneye, I get the following error:

```
Traceback (most recent call last):
  File "/usr/local/bin/meyectl", line 7, in <module>
    from motioneye.meyectl import main
  File "/usr/local/lib/python3.5/dist-packages/motioneye/meyectl.py", line 28, in <module>
    import settings
ImportError: No module named 'settings'
```
While, I probably just need to have a settings.py file somewhere, the installation instructions don't really cover this; I think there should be a note somewhere about it.

Thoughts? What's the right way to have my environment setup for this?

Maybe the Python install for OpenCV messed up motioneye?

jgwinner, 2018-11-03:
Ah! This is running on Python 3.5. Motioneye is still 2.7. Is that the issue?

I DO think it would be good to explain how to switch this in the install docs.

== John ==

jgwinner, 2018-11-04:
Hmm ... weirder and weirder:

```
root@raspberrypi:/usr/local/bin# stat /usr/bin/python
  File: /usr/bin/python -> python2.7

```

I'll install virtualenv, and reinstall everything.

== John ==

jgwinner, 2018-11-08:
The fix: I had forgotten I'd loaded virtualenv and switched to the Python 3 environment; even though the root path was still to python 2.7, pip ended up putting motioneye into python 3.

Not motioneye's issue (although it would be great if it ran on python 3 ...)

== John ==

closed 2018-11-08 by jgwinner

StefanRied, 2020-06-23:
sorry, don't get this. Have the same error. How to fix it? Thanks

zagrim, 2020-06-24:
jgwinner seems to have got rid of it by making sure to install MotionEye on Python 2.7 instead of 3. What's your exact error message? If it says anything about "python3", you need to either run all commands with `pip` replaced by `pip2`, or using other means of forcing everything to happen on 2.7 environment of Python.

jgwinner, 2020-06-24:
Exactly - more of a python problem than Motioneye. Honestly, I'm not using MotionEye anymore, so it isn't a problem; using all Python 3.x code under the right environment.

I hope the above and @zagrim's advice helps.

Basically, when you install Linux, there will be a mixture of Python 2.7 and Python 3 on your machine. Not all software will run on both versions. So, your system will have a way to run any "python" program on either 2.7 or on 3. Each Linux system is a little different (of course) so I can't give you specific advice.

In my case, it was to use virtualenv and switch back to python 2.7, then install MotionEye into the 2.7 environment. I'd read up on VirtualEnv.

        == John ==

StefanRied, 2020-06-24:
Thanks a lot. Yes, I had used pip3 instead of pip, as I wasn't aware of the compatibility problem.
Got it running now. I love MotionEye. Great software. I'll run it on a Raspberry PI Zero

ozett, 2020-07-21:
same problem with me, installed all on python 3.x
but the general install instructions should emphasise to keep it all in python 2.x (not only saying "use pip2"...) to make it clearer for the noobs like me.

https://github.com/ccrisan/motioneye/wiki/Installation#install-instructions

corbolais, 2021-09-03:
In fact, you *can* install motioneye on python3:

```
pip --version
pip 20.3.4 from /usr/lib/python3/dist-packages/pip (python 3.9)
pip install motioneye
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting motioneye
  Downloading https://www.piwheels.org/simple/motioneye/motioneye-0.42.1-py3-none-any.whl (792 kB)
Collecting pycurl
  Downloading https://www.piwheels.org/simple/pycurl/pycurl-7.44.1-cp39-cp39-linux_armv7l.whl (286 kB)
Collecting pillow
  Downloading https://www.piwheels.org/simple/pillow/Pillow-8.3.2-cp39-cp39-linux_armv7l.whl (1.1 MB)
Collecting tornado<6,>=3.1
  Downloading https://www.piwheels.org/simple/tornado/tornado-5.1.1-cp39-cp39-linux_armv7l.whl (457 kB)
Collecting jinja2
  Downloading https://www.piwheels.org/simple/jinja2/Jinja2-3.0.1-py3-none-any.whl (133 kB)
Collecting MarkupSafe>=2.0
  Downloading https://www.piwheels.org/simple/markupsafe/MarkupSafe-2.0.1-cp39-cp39-linux_armv7l.whl (25 kB)
Installing collected packages: MarkupSafe, tornado, pycurl, pillow, jinja2, motioneye
Successfully installed MarkupSafe-2.0.1 jinja2-3.0.1 motioneye-0.42.1 pillow-8.3.2 pycurl-7.44.1 tornado-5.1.1
```

The topic's error just remains.
No solution yet.

corbolais, 2021-09-03:
See also https://github.com/ccrisan/motioneye/issues/898

starbasessd, 2021-09-03:
@corbolais You do realize:
A\) you are commenting on very old _CLOSED_ issues.
B\) these comments are not reporting new issues, nor contributing to solving them?

corbolais, 2021-09-03:
> @corbolais You do realize:
> A) you are commenting on very old _CLOSED_ issues.
> B) these comments are not reporting new issues, nor contributing to solving them?

@starbasessd You do realize:
A) you're not contributing anything of substance.
B) the issue reported by me is still prevailing using a downloadable release?
C) There should rather be some serious effort to get all those motivated people working on preferrably one single branch to focus energy as opposed to what we have now, 3 ports, independent, non-functional and a complete waste of resources.

People like you are all over the place. Commenting on and harassing others without any reason and substance. I think there's a name for those entities, troll or something.