motioneye-project/motioneye #1050: second ethernet port for IP cameras

kmsgli, 2018-11-04:
Background:
Looking for some confirmation and a nudge in the correct direction.  Admittedly network stuff is not my thing but I can usually muddle my way through.  I have two IP cameras soon to be four running via Motioneye on my Arch Linux desktop computer.  Everything works fine but i loose footage occasionally due to piss poor internet service.  

Question:
Can I use a second ethernet port via PCIe and run the POE switch line back to my computer direct rather then the router so regardless of internet connection I will not loose video recordings.

If so what should i be searching to get the information on how to setup a second LAN that has no actual internet.

thanks in advanced.

jasaw, 2018-11-04:
Poor internet connection should not cause the cameras to lose connection to your desktop computer, unless you have an unreliable router and/or Ethernet switch, depending on your network setup.

You haven't provided detailed information on your network setup, but if you have a typical setup like this, I don't see why it wouldn't work if you're just recording straight on the desktop computer.
`Internet --- Router (DHCP server) --- PoE Ethernet Switch --- Desktop computer & all cameras connected directly to switch`

kmsgli, 2018-11-04:
I have it setup this way.
Modem to router then router to PC router to switch switch to cameras. So
the PC and the cameras are not directly connected both go back to the
router. Will connecting PC to switch instead of router change anything?

When my modem cuts out camera footage cuts out as well.

kmsgli, 2018-11-04:
> Yes, connecting your PC to the switch may change things, again depending
> on your equipment & config.
>
> Try to diagnose your network problem first.
>
>    - From a camera, try pinging your PC via IP address rather than
>    hostname, even when internet connection is not available.
>    - Try pinging again after connecting your PC to the switch.

The cameras have a rtsp address that I use to grab from motioneye. I am not
sure how to pin from the camera to the PC. Can I just use ping command from
the PC terminal and use cameradress to PC address?

jasaw, 2018-11-04:
Ah, yes, for your setup, PC pinging the camera makes more sense.

kmsgli, 2018-11-04:
I will ping both cameras tonight and report back in the morning as I'm at
work. I suppose I didn't think this could be an issue with my system rather
then my spotty internet. So in theory if the PC is attached to the switch
not the router I should be getting camera footage with the modem
disconnected?

jasaw, 2018-11-04:
Yes, assuming your router is at fault for not routing when internet goes down.

kmsgli, 2018-11-04:
Well then I have some testing to do. I have an Asus ac 1750. I'll have to
see what the story is with it. Thanks for pointing me in the right
direction direction. I will return with more information.

kmsgli, 2018-11-04:
So with the cable modem off i can still ping both cameras from the motioneye computer. Makes me think maybe network is not the issue.

The output of journalctl -u motioneye gives me this which is why I figured my corrupted videos and missing files are due to network

Nov 02 11:19:23 josh-desktop motion[5925]: [1:nc1] [ALR] [NET] netcam_handler_loop: Camera handler thread [5] started
Nov 02 11:19:25 josh-desktop motion[5925]: [1:ml1] [ERR] [ENC] get_oformat: ffmpeg_video_codec option value mp4:h264_omx is not supported
Nov 02 11:19:25 josh-desktop motion[5925]: [1:ml1] [ERR] [EVT] event_ffmpeg_newfile: ffopen_open error creating (new) file [/home/Media/Cameras/2018-11-02/11-19-25]: Operation now in progress
Nov 02 11:19:25 josh-desktop meyectl[5487]:    ERROR: connection problem detected for mjpg client for camera 1 on port 8081
Nov 02 11:19:25 josh-desktop motion[5925]: [1:nc1] [ALR] [NET] netcam_handler_loop: netcam camera handler: finish set, exiting
Nov 02 11:19:26 josh-desktop meyectl[5487]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov 02 11:19:26 josh-desktop meyectl[5487]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov 02 11:19:26 josh-desktop motion[5925]: [1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.1.200/11)
Nov 02 11:19:26 josh-desktop meyectl[5487]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov 02 11:19:26 josh-desktop meyectl[5487]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov 02 11:19:26 josh-desktop meyectl[5487]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov 02 11:19:27 josh-desktop meyectl[5487]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov 02 11:19:27 josh-desktop meyectl[5487]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov 02 11:19:27 josh-desktop meyectl[5487]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov 02 11:19:27 josh-desktop meyectl[5487]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov 02 11:19:27 josh-desktop meyectl[5487]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov 02 11:19:28 josh-desktop meyectl[5487]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov 02 11:19:28 josh-desktop meyectl[5487]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov 02 11:19:28 josh-desktop meyectl[5487]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov 02 11:19:28 josh-desktop meyectl[5487]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov 02 11:19:28 josh-desktop meyectl[5487]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov 02 11:19:29 josh-desktop meyectl[5487]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov 02 11:19:29 josh-desktop meyectl[5487]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov 02 11:19:29 josh-desktop meyectl[5487]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov 02 11:19:29 josh-desktop meyectl[5487]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov 02 11:19:29 josh-desktop meyectl[5487]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov 02 11:19:30 josh-desktop meyectl[5487]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov 02 11:19:30 josh-desktop meyectl[5487]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov 02 11:19:30 josh-desktop meyectl[5487]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov 02 11:19:30 josh-desktop meyectl[5487]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov 02 11:19:30 josh-desktop meyectl[5487]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov 02 11:19:31 josh-desktop meyectl[5487]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov 02 11:19:32 josh-desktop meyectl[5487]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov 02 11:19:32 josh-desktop motion[5925]: [1:nc1] [ALR] [NET] netcam_handler_loop: Camera handler thread [5] started
Nov 02 11:19:34 josh-desktop motion[5925]: [1:ml1] [ERR] [ENC] get_oformat: ffmpeg_video_codec option value mp4:h264_omx is not supported
Nov 02 11:19:34 josh-desktop motion[5925]: [1:ml1] [ERR] [EVT] event_ffmpeg_newfile: ffopen_open error creating (new) file [/home/Media/Cameras/2018-11-02/11-19-34]: Operation now in progress
Nov 02 11:19:34 josh-desktop meyectl[5487]:    ERROR: connection problem detected for mjpg client for camera 1 on port 8081
Nov 02 11:19:34 josh-desktop motion[5925]: [1:nc1] [ALR] [NET] netcam_handler_loop: netcam camera handler: finish set, exiting
Nov 02 11:19:34 josh-desktop meyectl[5487]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov 02 11:19:34 josh-desktop motion[5925]: [1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.1.200/11)
Nov 02 11:19:34 josh-desktop meyectl[5487]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov 02 11:19:34 josh-desktop meyectl[5487]:  WARNING: Connect error on fd 15: ECONNREFUSED
Nov 02 11:19:35 josh-desktop meyectl[5487]:  WARNING: Connect error on fd 15: ECONNREFUSED

jasaw, 2018-11-04:
I can see two problems from the log.

>Nov 02 11:19:25 josh-desktop motion[5925]: [1:ml1] [ERR] [ENC] get_oformat: ffmpeg_video_codec option value mp4:h264_omx is not supported

You have configured your PC to use h264_omx to encode videos, but that's not supported on your platform. It'll just fallback to the default encoder, won't cause video corruption issue that you're seeing. This is just FYI.

>Nov 02 11:19:25 josh-desktop motion[5925]: [1:ml1] [ERR] [EVT] event_ffmpeg_newfile: ffopen_open error creating (new) file [/home/Media/Cameras/2018-11-02/11-19-25]: Operation now in progress

This looks like the video recording problem that you're having. What's your ffmpeg version? 
I think motion developers might be able to help you. To report a motion issue, follow this guide: https://github.com/ccrisan/motioneye/wiki/Report-Motion-Issue

kmsgli, 2018-11-05:
I am sorry that log is from days ago I did figure out the issue with the wrong video format and fixed it tto mpeg. 

here is a current output:

ov 04 13:39:34 josh-desktop meyectl[798]:  WARNING: Connect error on fd 13: ECONNREFUSED
Nov 04 13:39:34 josh-desktop meyectl[798]:  WARNING: Connect error on fd 13: ECONNREFUSED
Nov 04 13:39:35 josh-desktop meyectl[798]:  WARNING: Connect error on fd 13: ECONNREFUSED
Nov 04 13:39:35 josh-desktop meyectl[798]:  WARNING: Connect error on fd 13: ECONNREFUSED
Nov 04 13:39:35 josh-desktop meyectl[798]:  WARNING: Connect error on fd 13: ECONNREFUSED
Nov 04 13:39:35 josh-desktop meyectl[798]:  WARNING: Connect error on fd 13: ECONNREFUSED
Nov 04 13:39:35 josh-desktop meyectl[798]:  WARNING: Connect error on fd 13: ECONNREFUSED
Nov 04 13:39:35 josh-desktop meyectl[798]:  WARNING: Connect error on fd 13: ECONNREFUSED
Nov 04 13:39:35 josh-desktop meyectl[798]:  WARNING: Connect error on fd 13: ECONNREFUSED
Nov 04 13:39:35 josh-desktop meyectl[798]:  WARNING: Connect error on fd 13: ECONNREFUSED
Nov 04 13:39:35 josh-desktop motion[3607]: [2:nc2] [ALR] [NET] netcam_handler_loop: Camera handler thread [4] started
Nov 04 13:39:35 josh-desktop meyectl[798]:  WARNING: Connect error on fd 13: ECONNREFUSED
Nov 04 13:39:35 josh-desktop meyectl[798]:  WARNING: Connect error on fd 19: ECONNREFUSED
Nov 04 13:39:35 josh-desktop meyectl[798]:  WARNING: Connect error on fd 19: ECONNREFUSED
Nov 04 13:39:35 josh-desktop meyectl[798]:  WARNING: Connect error on fd 19: ECONNREFUSED
Nov 04 13:39:35 josh-desktop motion[3607]: [1:nc1] [ALR] [NET] netcam_handler_loop: Camera handler thread [5] started
Nov 04 14:09:53 josh-desktop meyectl[798]:    ERROR: failed to create movie preview for /home/Media/Cameras/2018-11-04/13-39-37.mp4: Command '[u'ffmpeg', u'-i', u'/home/Media/Cameras/2018-11-04/13-39-37.mp4', u'-f', u'mjpeg', u'-vframes', u'1', u'-ss', u'4', u'-y', u'/home/Media/Cameras/2018-11-04/13-39-37.mp4.thumb']' returned non-zero exit status 69
Nov 04 14:09:55 josh-desktop meyectl[798]:    ERROR: failed to create movie preview for /home/Media/Cameras/2018-11-04/13-39-37.mp4: Command '[u'ffmpeg', u'-i', u'/home/Media/Cameras/2018-11-04/13-39-37.mp4', u'-f', u'mjpeg', u'-vframes', u'1', u'-ss', u'4', u'-y', u'/home/Media/Cameras/2018-11-04/13-39-37.mp4.thumb']' returned non-zero exit status 69
Nov 04 14:12:14 josh-desktop meyectl[798]:    ERROR: failed to create movie preview for /home/Media/Cameras/2018-11-04/14-09-37.mp4: Command '[u'ffmpeg', u'-i', u'/home/Media/Cameras/2018-11-04/14-09-37.mp4', u'-f', u'mjpeg', u'-vframes', u'1', u'-ss', u'4', u'-y', u'/home/Media/Cameras/2018-11-04/14-09-37.mp4.thumb']' returned non-zero exit status 1
Nov 04 14:13:09 josh-desktop meyectl[798]:    ERROR: failed to create movie preview for /home/Media/Cameras/2018-11-04/11-23-09.mp4: Command '[u'ffmpeg', u'-i', u'/home/Media/Cameras/2018-11-04/11-23-09.mp4', u'-f', u'mjpeg', u'-vframes', u'1', u'-ss', u'4', u'-y', u'/home/Media/Cameras/2018-11-04/11-23-09.mp4.thumb']' returned non-zero exit status 1
Nov 04 14:13:09 josh-desktop meyectl[798]:    ERROR: failed to create movie preview for /home/Media/Cameras/2018-11-04/11-05-39.mp4: Command '[u'ffmpeg', u'-i', u'/home/Media/Cameras/2018-11-04/11-05-39.mp4', u'-f', u'mjpeg', u'-vframes', u'1', u'-ss', u'4', u'-y', u'/home/Media/Cameras/2018-11-04/11-05-39.mp4.thumb']' returned non-zero exit status 1
Nov 04 14:13:29 josh-desktop meyectl[798]:    ERROR: failed to create movie preview for /home/Media/Cameras/2018-11-04/14-09-37.mp4: Command '[u'ffmpeg', u'-i', u'/home/Media/Cameras/2018-11-04/14-09-37.mp4', u'-f', u'mjpeg', u'-vframes', u'1', u'-ss', u'4', u'-y', u'/home/Media/Cameras/2018-11-04/14-09-37.mp4.thumb']' returned non-zero exit status 1
Nov 04 14:13:30 josh-desktop meyectl[798]:    ERROR: failed to create movie preview for /home/Media/Cameras/2018-11-04/11-23-09.mp4: Command '[u'ffmpeg', u'-i', u'/home/Media/Cameras/2018-11-04/11-23-09.mp4', u'-f', u'mjpeg', u'-vframes', u'1', u'-ss', u'4', u'-y', u'/home/Media/Cameras/2018-11-04/11-23-09.mp4.thumb']' returned non-zero exit status 1
Nov 04 14:13:30 josh-desktop meyectl[798]:    ERROR: failed to create movie preview for /home/Media/Cameras/2018-11-04/11-05-39.mp4: Command '[u'ffmpeg', u'-i', u'/home/Media/Cameras/2018-11-04/11-05-39.mp4', u'-f', u'mjpeg', u'-vframes', u'1', u'-ss', u'4', u'-y', u'/home/Media/Cameras/2018-11-04/11-05-39.mp4.thumb']' returned non-zero exit status 1
Nov 04 14:20:16 josh-desktop motion[3607]: [1:nc1] [ERR] [ENC] ffmpeg_avcodec_log: Invalid NAL unit 12, skipping.
Nov 04 15:09:44 josh-desktop meyectl[798]:    ERROR: failed to create movie preview for /home/Media/Cameras/2018-11-04/14-39-37.mp4: Command '[u'ffmpeg', u'-i', u'/home/Media/Cameras/2018-11-04/14-39-37.mp4', u'-f', u'mjpeg', u'-vframes', u'1', u'-ss', u'4', u'-y', u'/home/Media/Cameras/2018-11-04/14-39-37.mp4.thumb']' returned non-zero exit status 69
Nov 04 15:09:46 josh-desktop meyectl[798]:    ERROR: failed to create movie preview for /home/Media/Cameras/2018-11-04/14-39-37.mp4: Command '[u'ffmpeg', u'-i', u'/home/Media/Cameras/2018-11-04/14-39-37.mp4', u'-f', u'mjpeg', u'-vframes', u'1', u'-ss', u'4', u'-y', u'/home/Media/Cameras/2018-11-04/14-39-37.mp4.thumb']' returned non-zero exit status 69
Nov 04 15:48:10 josh-desktop motion[3607]: [1:nc1] [ERR] [ENC] ffmpeg_avcodec_log: Invalid NAL unit 12, skipping.
[... cut ...] , u'-f', u'mjpeg', u'-vframes', u'1', u'-ss', u'4', u'-y', u'/home/Media/Cameras/2018-11-04/19-39-37.mp4.thumb']' returned non-zero exit status 69
Nov 04 20:09:47 josh-desktop meyectl[798]:    ERROR: failed to create movie preview for /home/Media/Cameras/2018-11-04/19-39-37.mp4: Command '[u'ffmpeg', u'-i', u'/home/Media/Cameras/2018-11-04/19-39-37.mp4', u'-f', u'mjpeg', u'-vframes', u'1', u'-ss', u'4', u'-y', u'/home/Media/Cameras/2018-11-04/19-39-37.mp4.thumb']' returned non-zero exit status 69

kmsgli, 2018-11-05:
> Just by looking at your log, I can see two problems, but I don't know what
> caused them and how to solve them.
> The camera timed out here, which explains why you're losing video feed.
> This is starting to look like motion software problem, and motion
> developers might be able to help you. See my earlier post on how to report

Thanks for your input. I'll start poking around with motion directly and
see if I can get some help on the motion boards.

Thanks again.

JamBalaya56562, 2026-11-08:
Closing, as the network question was answered above: you could still ping both cameras with the modem off, so the lost footage was not caused by the internet connection. The errors left in the log, which jasaw pointed out, were to be followed up with Motion, as you said. Please open a new issue with `motion.log` if recordings still get lost with current releases.

closed 2026-11-08 by JamBalaya56562