motioneye-project/motioneye #1051: CPU Question

rdavey228, 2018-11-04:
Hi

I have 4 1080p PoE IP cameras running into MotionEye.

To start with I had the resolution at 1920x1080 and for motion detection and image grabs on motion detection this was fine.

My problem starts when I want to record video! At this resolution, even with only 1 camera enabled (the rest are off) my CPU is 100% all the time. I have tried to drop the resolution of the camera down to 1280x768 but CPU is still 100% when recording and FPS drops from 8/10fps to 0.5/1fps 2fps if I'm lucky.

The cameras are running into a top of the line Cisco POE gigabit switch and are on their own dedicated CCTV VLAN along with MotionEye.

I am running motioneye inside a docker container. The docker installation sits inside a Hyper-V VM which is running Ubuntu 18.04. The Underlying Ubuntu Installation that runs docker has 8vCPU assigned to it and 8GB of RAM. Nothing else is running on this ubuntu VM apart from Docker and motioneye is the only container that Docker is running.

Any help would be appreciated please?

Thanks
Ross

jasaw, 2018-11-05:
Try passthrough recording. 
https://rawgit.com/Motion-Project/motion/master/motion_config.html#movie_passthrough

rdavey228, 2018-11-05:
Thanks but that doesn’t help me

The documentation says it will ignore any text overlays which I need and also any masks applied to the camera. I need the masks as it’s in a residential area and part of the camera covers a road that I do not want to trigger on every time a car goes past. 

I can’t believe that even with one camera that at 640x960 resolution can’t record without spiking the cpu to 100% on a system that has been given 6 processors and is on enterprise spec hardware that’s just ludicrous

jasaw, 2018-11-07:
Your RTSP video stream is already encoded. To do motion detection on it, your server needs to decode (in CPU) the video stream, then run motion detection on the raw video. Motion software then adds text overlays to the raw video, then encodes it (in CPU) to whatever format that you've selected. The video encoding is VERY CPU intensive.

With pass-through recording, the idea is to skip the encoding part. That also means you can't add text overlays onto the video because that involves decode-add-encode.

If you still want text overlays, you could offload video encoding (and maybe encoding too) to the GPU. Make sure your ffmpeg library supports hardware accelerated video encoder. In your `thread-1.conf` file, set your preferred encoder this way `ffmpeg_video_codec <format>:<encoder>`. Example `ffmpeg_video_codec mp4:h264_nvenc`. Which encoder to use depends on what's available on your system. Please read [ffmpeg hwaccel documentation](https://trac.ffmpeg.org/wiki/HWAccelIntro) for more info. If you do go down this option, please report back on how you went. I'm interested to know how hwaccel perform on other systems.

rdavey228, 2018-11-08:
Hi jasaw

I have tried your suggestion but my CPU is still flatlined at 100%.

Only happens when recording. My CPU when streaming but not recording all 4 cameras at 1920x1080 is hovering around 20/30% CPU.

jasaw, 2018-11-09:
That means you're still using CPU to encode the video. Make sure you actually have a hardware accelerated video encoder.
Run `ffmpeg -encoders` and see which hardware accelerated video encoder is available on your system. `h264_nvenc` won't be available if you don't have nVidia GPU. If you have Intel GPU, try `h264_vaapi` instead if it's available.

rdavey228, 2018-11-10:
Ok, I tried something different and installed motioneye onto its own box. This was a spare Dell R610 I had lying around. The system has 72GB of memory and 12 cores to which I assigned all of it to this system!

I set my camera at the lowest resolution and set the camera to record. 

Turns out whilst I can use GPU acceleration, the GPU in these machines is very basic so wouldst be much use so I'm back to using CPU.

CPU still spiked out at 100%....I cannot understand that with all that power the system still cant keep up recording a single camera at such a small resolution. I am currently recording all of my cameras at full resolution into a windows OS based NVR at the moment all with motion detection and text and zone overlays with only 4 CPUs assigned and 8gb of memory and it doesn't bat an eyelid....so why is motion so bad at doing utilising system resources when I've given it the entire 12 cores of the CPU and 72GB of memory.

I wanted to move over to motioneye as its a smaller footprint being able to run it on a core Ubuntu OS inside docker rather than spinning up an entire windows environment. I also like the web browser of Motioneye for viewing my cameras too, something my current NVR won't do unless I pay for it as its not included in the free version.

jasaw, 2018-11-11:
I see you have these hardware accelerated encoders, but don't know why your CPU is still running at 100%
> V..... h264_nvenc           NVIDIA NVENC H.264 encoder (codec h264)
 V..... h264_omx             OpenMAX IL H.264 video encoder (codec h264)
 V..... h264_vaapi           H.264/AVC (VAAPI) (codec h264)

You could try using ffmpeg to transcode a video using one of these encoders, see what your CPU load is like. If your CPU load is still 100%, then there's nothing motionEye or motion can do about it.

About the passthrough recording, you could try doing motion detection on the low res video, and record the high res video. Most IP cameras support 2 video streams (low and high res). You might also want to ask [motion](https://github.com/Motion-Project/motion) developers about your passthrough recording using 100% CPU.

kleini, 2018-11-12:
See https://github.com/ccrisan/motioneye/blob/master/motioneye/mediafiles.py#L46

There is a hard selection of codecs, which can not be replaced easily using the accelerated encoders.

jasaw, 2018-11-12:
@kleini Yes, the codec selection cannot be changed to an arbitrary codec from motionEye front end, but can be changed by editing the thread-1.conf file directly.

SealedJoy, 2018-11-12:
+1 also experiencing mad cpu load on ubuntu desktop (mint) install. I am using a single usb webcam at 640x480 (microsoft lifecam 5000) with motion detection off and still hitting 90% cpu with 8gb mem

jkaberg, 2018-11-27:
I'd just like to mention that if you're using the docker container and want to do GPU stuff, you need to pass the GPU into the container, eg;

`docker run ... --device /dev/dri:/dev/dri --privileged ...`

https://docs.docker.com/engine/reference/commandline/run/#add-host-device-to-container---device

nopmop, 2019-02-24:
I don't think it's related (only) to encoding.
On my RPi 3 B+ the CPU is 100% busy only when the framerate is over 4, which coincidentally is the number of processors.

Could it be a spinlock issue between threads??

maxi1134, 2019-12-25:
Did you ever found a solution to this issue?

Joshua2504, 2020-01-12:
I have to leave a comment on this.

I'm currently setting up a motioneye environment on an home server running 8 GB of memory and an i5-7400. Running 1x 1080p, 2x 1440p, 2x 360p. No recording, no motion detection. The CPU burns to death and the rtsp streams are barely smooth at just 10fps!

Streams are coming over a dedicated wifi bridge (cameras in it's own vLAN, behind a powerfull Cisco PoE switch too...), so I thought it's related to that somehow, but running all streams via VLC works just fine.

Shinobi and others can run just fine while recording, and it's a shame motion(eye). can't even without.

kleini, 2020-01-13:
@Joshua2504 Maybe its time to drop the motioneye project and use something else, if it is better. Why do you stick to motioneye and blame this community project?

Joshua2504, 2020-01-13:
@kleini motioneye builds on motion which is the issue at this point. motion is a CPU-roasting monster as soon as it starts to touch your raw streams in any way. motioneye was made for a raspberry pi and motion alone can't even really run on "good" state-of-the-art hardware.

I'm pretty happy with Shinobi now (setup given above, 30% CPU usage while permanent recording on all cameras and 2x motion detection on 1440p), although it has kinda active development. I can only excuse for blaming the motioneye community on this, as this is not in your hands at all. you'ld need to build your own ffmpeg stuff and ditch motion to get the CPU load under control. If I would use my Ryzen 2600x (or a dedicated GPU with HW encoding) to decode and encode one 1440p 30fps rtsp H256 with motion(eye), it would probably run just fine.

kleini, 2020-01-13:
motion can use HW-acceleration, which would avoid CPU-roasting. But HW may be very different and HW video support in Linux is hard: VDPAU, VA-API and vendor specific APIs, HW encoding is even harder. I was never able to get good hardware acceleration working under Linux.
motioneye is a Python project, it is not made for Raspberry Pi. It works very well everywhere, but motion is the not so easy thing.

JamBalaya56562, 2026-11-10:
Closing, as this was answered: Motion decodes every frame of the stream on the CPU to detect motion and encodes it again for movies, which jasaw and kleini explained. Movie Passthrough skips the encoding, a hardware encoder can take it over where ffmpeg has one, and detecting on the camera's low-resolution substream lowers the decoding load.

closed 2026-11-10 by JamBalaya56562