motioneye-project/motioneye #1053: Raspberry Pi motionEye (on Raspbian) high CPU usage

mxplea10, 2018-11-06:
Reference: motionEye ([Install on Raspbian](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian))

On a Raspberry Pi 3 B+, I installed motionEye on Raspbian.  With only three (3) cameras (Dahua IP cameras) added to motionEye via the RTSP url, the Raspberry Pi's CPU usage fluctuates between 77 and 82%.  With four (4) or more IP cameras added, the Raspberry Pi's CPU usage is 100%.  All motionEye settings are still set at default.  When the motionEye server is stopped, the Raspberry Pi's CPU usages fluctuates between 0 and 5%.  Are there setting changes in the motionEye front end or on the Raspberry Pi that can reduce the load that the motionEye server puts on the Raspberry Pi's CPU?  Is there a maximum number of cameras for motionEye?

Also, is there a proper way to log out of the motionEye's frontend instead of just closing the browser?

[image]

SealedJoy, 2018-11-12:
You almost certainly need more powerful hardware.
I would recommend switching off motion detection, noise detection and stream resizing for as many cameras as possible, use the frame rate/ res dimmer. You could also turn off video recording for as many cameras as possible, perhaps replacing with the take photos option. this will consume alot less cpu.

mxplea10, 2018-11-12:
I totally agree that a more powerful hardware platform is needed.  My Raspberry Pi collection is growing and I've like to find an alternate small footprint hardware platform that I will run multiple applications.  I've reduced the load significantly by viewing the video sub stream instead of the main stream of the cameras.  Thanks for your input and advice.

closed 2018-11-16 by mxplea10

mxplea10, 2018-11-16:
I'll test this for while to determine it's best use for me.  Initially, I only wanted to view 2 to 4 cameras but decided to check out the video feature and saving the video to a network shared folder.