motioneye-project/motioneye #1055: cant install motioneyeos on raspberry pi 3 b+

simo174s, 2018-11-07:
i have tryed to install motioneyeos on my raspberry, without success.
this is what i have done.
i have flashed my usb drive using etcher and also wind 32 disk imager.
i have used the .img "motioneyeos-raspberrypi3-20180627"
i insert the flashed usb in the raspberry, i connect the lan cable, the hdmi cable, the keyboard, and i switch on raspberry. 
raspberry start, the multicolor screen appear,
and then the rasperry is still locked on the black screen with 4 raspberry on the top left of the screen. and under the 4 raspberry there is a "-" that blink all time and nothing more...
can you help me? thanks

jasaw, 2018-11-07:
looks like your raspberry pi didn't manage to see the `boot` partition. A few things for you to try:
1. Insert the SD card to your PC, see whether you can see the contents in `boot` partition. There should be file like `config.txt`, `cmdline.txt`, `bootcode.bin`, `fixup.dat`... If you don't see those files, that means your SD card is corrupted.
2. Flash Raspbian, see if it boots up.
3. Try a different SD card.

jasaw, 2018-11-08:
After flashing motionEyeOS onto the SD card, plug the SD card into your PC. Do you see the `boot` partition? Do you see files like `config.txt`, `cmdline.txt`, `bootcode.bin`, `fixup.dat` in the `boot` partition?

electron1979, 2018-12-10:
No reply?
Close this issue!

ccrisan, 2018-12-15:
Closing due to no reply from OP.

closed 2018-12-15 by ccrisan

nirvanasinha, 2019-03-30:
I am having the Same issue, and i have checked all the files but its not working.
please help ?