motioneye-project/motioneye #1056: USB and menu problem

stonefox1958, 2018-11-09:
MotionEye - need help

Hi
Looking after help "MotionEye"

**1:**
How do I get inside this menu? Have tried "PuTTY" logged in and written "Sudo raspi-config" but it does not work
Gets this error "-sh: sudo command not found"

**2**:
I have put a Logitech 9000 Webcam into one of the USB ports, but the camera will not appear when I try to add it.
Have watched some Youtube videos and see that others have an option called "USB camera", but there is no one among the listed in the dropdown menu.

**3:**
The left side shows the options I have in "MotionEye" ( a raspberri org. camera ) connectors directly on the board, but how do I get the more advanced menu shown?

[image]

ccrisan, 2018-11-09:
1. You're on motionEyeOS, not motionEye; see https://github.com/ccrisan/motioneyeos/wiki/FAQ#why-doesnt-sudo-work

2. Does it work on Raspbian?

3. Just enable Advanced Options in the Main section.

JamBalaya56562, 2026-11-08:
Closing as answered by ccrisan above: this was motionEyeOS, where `sudo` is not needed (see its FAQ), and the more advanced settings show up once Advanced Options is enabled in the General section. Questions about motionEyeOS itself belong in its own repository (motioneye-project/motioneyeos).

closed 2026-11-08 by JamBalaya56562