motioneye-project/motioneye #1057: Captured Before not working as expected?

olbasoil, 2018-11-09:
My scenario is that I want to trigger a recording from a gate sensor in Home Assistant, 
curl "http://[motion eye ip]:7999/1/config/set?emulate_motion=1 starts a recording and emulate_motion=0 stops the recording. However there is a very slight lag, and I would ideally like to start the recording from a second or two before the sensor is activated,
I have set a 'captured before' to 60 frames. But this does not seem to have any effect? Is that because it is 'emulated' motion rather than a real event?

I have tried different values in the 'captured before' box, but noticed no change. If it did work, then do I take the frame rate into account when calculating frames? (ie 10 frames @ 5fps = 2 secs, but at 1 fps = 10secs?)

I am using 
motionEye Version  0.39.2
Motion Version  4.1.1
OS Version  motionEyeOS 20180627

jasaw, 2018-11-12:
From the motion source code, it looks like emulate motion does not take `Captured Before` into account. To support `Captured Before` in emulate motion mode is trivial, but you'll need to get [motion](https://github.com/Motion-Project/motion) developers to agree to it first.

Yes, for `Captured Before`, you need to take frame rate into account.

JamBalaya56562, 2026-11-09:
Checked with Motion 4.7.1 at 10 fps: a 5 s event started with `emulate_motion=1` gave 50 frames with "Captured Before" 0 and 81 frames with 30, so the frames before the trigger are kept now. The same holds for the "start a motion event" button / `POST /action/<camera_id>/eventstart` (#2691). Closing, as this works with current Motion.

closed 2026-11-09 by JamBalaya56562