motioneye-project/motioneye #1062: Webhook replaces Command

dgomes, 2018-11-12:
**Running:**
motionEye Version | 0.39.3
Motion Version | 4.1.1
OS Version | Ubuntu 18.04

**Issue**
Whenever I setup "Run a command" with a command such as:
```
curl -X POST -F "image=@%f" http://192.168.1.1/api/webhook/crazy_token
```

Motioneye disables my command and moves it into the web hook URL. Unfortunantely the webhook will not POST my image.

**Expected behaviour**
Leave "Run a command" alone

ccrisan, 2018-11-16:
@dgomes you're right; in fact, what motionEye does is to look for the word "webhook" in the list of commands to execute and, if found, considers that the command is in fact a webook configuration. We need to revise this and find a better way to tell regular commands from webooks.

dgomes, 2018-11-16:
Can't this "automation" be simply disabled ? What was the initial purpose ?

ccrisan, 2018-11-16:
@dgomes Motion exposes an option called e.g. `on_movie_end` which represents a command to be executed when a movie is done. Now, in order to accommodate both webhooks and custom commands (and possibly other things, such as notifications), motionEye specifies multiple subcommands to this option. Identifying which one of them is the webhook is something not quite straight forward :)

dgomes, 2018-11-16:
@ccrisan thank you for your insights :)

in the current implementation, motioneye will simply call webhook and command one after the other, as commands as separated by ; in the on_movie_end call.

To overcome this issue I've created a no operation webhook (NOOP), but I don't see why MotionEye couldn't simply remove the webhook call when no webhook is defined. Am I missing something ?

ccrisan, 2018-11-16:
@dgomes the problem is the presence of the word "webhook" in your URL. For now you can try change that to something else and it will work.

IVData, 2019-02-20:
Having the same issue here on motioneye in docker. @dgomes solution of a dummy webhook entry doesn't work for me either - motioneye disables my command no matter what.

Wouldn't a good solution be to enable motioneye to upload files with the webhook option?

IVData, 2019-02-23:
For the prosperity of the internet, you can solve this by inserting a backslash in the middle of "webhook". Motion doesn't move the command to a webhook, and the backslash is escaped at some point before the command is run. My command for home assistant became:

`curl -X POST -F "image=@%f" http://my.ha.ip:8123/api/webhoo\k/my_whook_id`

freekatzi, 2019-02-28:
With your solution now the image from motioneye is still working.
In Configuration.yaml work this for me.

- platform: push
    name: Haustuerklingel_Einfahrt
    webhook_id: WH_Einfahrt_Haustuer_01
    buffer: 2

Now i configure a appdaemon dashboard where i need the image also.
But this dosent work. In dash file i try different settings but no image / video will be visible.
camera_front_door:
    widget_type: camera.push
    entity_url: xyz
    entity_picture: xyz

on widget_type the different posible  camera entry like mjpeg / push does not work
i try as example generic camera. Here it was possible to get a image if i dont use login credentials.
http://aaa.bbb.ccc.ddd:8765/picture/1/current/?_=1551328564666
With this as user credentials no image will show.
http://aaa.bbb.ccc.ddd:8765/picture/1/current/?_=1551328564666&_username=username&_signature=signaturevalue

The camera.push option get an image, so this image is available in system.
How can it be transfer to the dash?

And how it is possible to debug appdaemon to see what happend?

i hope my description is readable and will hold enough Informations for your to help.

I dont find at www more information for motioneye and appdaemon using push.
maybe you have a link?

Thank you for your assistant.
Regards Katzi

Cooper-Dale, 2019-02-28:
If your HA is running on https it  will not display frame from http. So try 
http**s**://aaa.bbb.ccc.ddd:8765/picture/1/current/?_=1551328564666

JamBalaya56562, 2026-11-08:
Closing as fixed by #1245 in 0.40: motionEye took any command containing the word `webhook`, such as a Home Assistant URL, for its own web hook. Since then it only matches ` webhook ` with spaces around it, as in the command it writes itself, so a custom command with `/api/webhook/` in its URL stays under Run a command.

closed 2026-11-08 by JamBalaya56562