motioneye-project/motioneye #1066: Feature request: S3 Upload Functionality

adamtbeames, 2018-11-14:
Hi @ccrisan,

I think MotionEye is a great project (thanks!), and I'm hoping to use it as the frontend for a new RPi based product I'm currently developing.
A requirement for one of our primary use cases is support for uploads to AWS S3, which is currently not available, and from what I can see not on the roadmap either.

I found one other [issue](https://github.com/ccrisan/motioneye/issues/364) requesting this feature,  but it was closed by the requester without a response from maintainers.

I've created a [PR](https://github.com/ccrisan/motioneye/pull/1063) adding basic functionality (albeit partly hardcoded) to show how this might be integrated.

Please let me know your thoughts.

Cheers,

Adam

MourIdri, 2018-11-14:
In that case I will if possible,  to add Azure and Google storage support.

adamtbeames, 2018-11-14:
@MourIdri thanks for the quick response.

My experience is mostly with AWS services, but having all three would be great.
I'm planning on integrating some other AWS services too, so it might be worth keeping that in the mind.  Happy to discuss with you further at a later stage.

Thanks again :-)

jkaberg, 2018-11-16:
You could script (using motionEye's "File Storage->Run a command") this to copy the file using [rclone](https://github.com/ncw/rclone) to an rclone supported backend (S3, Azure, GSS and many more..) if you only need to store it.

Example:
`/path/to/rclone copy %f googledrive:/motioneye`

and if you need delete the orginal file just add the following to above command 
`; rm -f %f`

adamtbeames, 2018-11-16:
Thanks I'll take a look.  I'd really like the functionality to be
configurable from the web interface.  The product were designing isn't
targeted at technical people per say.

ccrisan, 2018-11-16:
@adamtbeames thanks for your contribution. Having out-of-the-box S3 upload support in motionEye is a cool feature that many users could benefit from. Let's discuss more about this on the PR itself.

adamtbeames, 2018-11-16:
Thanks @ccrisan.  I know the code in the PR isn't ideal, but not being much of a python guy thats all I could muster :-)

MichaIng, 2022-04-06:
Implemented for next release/current `dev` branch: #2376
Would be great if you could test it, here are `dev` install instructions: https://github.com/motioneye-project/motioneye/tree/dev#installation

closed 2022-04-06 by MichaIng