motioneye-project/motioneye #1067: YUV444/mp4 codec unsupported in Firefox

jkaberg, 2018-11-16:
The YUV444/mp4 (aka "H.264 (.mp4) in motionEye) codec is unsupported [in Firefox](https://bugzilla.mozilla.org/show_bug.cgi?id=1368063#c4). An codec which works across all platforms (VP8/VP9) would be nice.

> Media resource https://XXXX/2018-11-16/05-42-03.mp4?_username=XXXX&_signature=XXXX could not be decoded, error: Error Code: NS_ERROR_DOM_MEDIA_FATAL_ERR (0x806e0005)
Details: mozilla::SupportChecker::AddMediaFormatChecker(const mozilla::TrackInfo&)::<lambda()>: Decoder may not have the capability to handle the requested video format with YUV444 chroma subsampling.

jkaberg, 2018-12-02:
For now I've just hacked around this with a simple conversion script 

Under File Storage -> Run a command
`/etc/motioneye/convert.sh %f`

convert.sh:
```
#!/bin/bash

FILEPATH="$1"
FILENAME="${FILEPATH##*/}"
TMP_PATH=/tmp/"$FILENAME".webm

/usr/bin/ffmpeg -i "$FILEPATH" -f webm -c:v libvpx -c:a libopus "$TMP_PATH"
/bin/rm -f "$FILEPATH"
/bin/mv "$TMP_PATH" "$FILEPATH"
```

jasaw, 2019-09-02:
YUV444 doesn't sound right. Motion sends images in YUV420 format to ffmpeg to encode. Please report this issue to [motion project](https://github.com/Motion-Project/motion) so they can give us some assistance.

gusarg81, 2020-08-31:
I am running with this same problem. But it does play on Chrome (does not in Firefox).

JamBalaya56562, 2026-11-07:
Closing as a duplicate of #1206: at 100% movie quality, Motion writes lossless H.264 with a 4:4:4 profile, which Firefox cannot play. Checked with Motion 4.7.1 on a Raspberry Pi: 100% gives `High 4:4:4 Intra`, 75% gives `Constrained Baseline`. Lowering the movie quality gives files that play, and with movie passthrough the format is the camera's own, as zagrim noted above.

closed 2026-11-07 by JamBalaya56562