motioneye-project/motioneye #1069: motioneye.conf.sample: No such file or directory during copy

VictorTagayun, 2018-11-18:
```
pi@raspberrypi:~ $ sudo cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
cp: cannot stat '/usr/local/share/motioneye/extra/motioneye.conf.sample': No such file or directory
```

should be 

`sudo cp /home/pi/.local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf`

as per 

[https://github.com/ccrisan/motioneye/issues/748](url)

please update your wiki

ccrisan, 2018-11-18:
@VictorTagayun the wiki clearly states that you need to run stuff as root (using `sudo`). If you install it as your own user, you'll need to adjust all commands accordingly.

closed 2018-11-18 by ccrisan

bmolyneaux, 2019-12-09:
Make sure you run `pip install motioneye` as root.

randynose, 2019-12-22:
I think that the issue is trying to copy the file before it's created. I was able to copy it once I did the pip install.

B0rax, 2020-11-13:
I did run the pip install command as root and it is not finding the file in the specified directory. The directory is not even created.

starbasessd, 2020-11-13:
Please follow the new directions for Ubuntu 20.04 located here:
https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29
as more and more Linux Distributions are disabling functionality of python
2.7 (and the associated programs) upstream, in anticipation of dropping all
support in the repositories for 31 Dec 2020.
Let me know if it works for you

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

angadsingh, 2020-12-02:
i did `pip install motioneye` as root and still faced the above issue

starbasessd, 2020-12-02:
Did you run the instructions I pointed you to? There are issues with the
upstream packages since python 2.7 and all python 2 packages are going
away. It defaults to python 3 and pip3 when you just run pip.
https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29
I would suggest starting at the beginning (no need to start from scratch
with a new image) and run through these instructions.

pip2 install motioneye

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mbrivio, 2021-01-25:
Trying to install motioneye on fresh and updated ubuntu 20.04 VM following the instruction above but when I need to install pip2 (sudo python2 get-pip.py) I get an error:
```
Traceback (most recent call last):
  File "get-pip.py", line 24226, in <module>
    main()
  File "get-pip.py", line 199, in main
    bootstrap(tmpdir=tmpdir)
  File "get-pip.py", line 82, in bootstrap
    from pip._internal.cli.main import main as pip_entry_point
  File "/tmp/tmpWF8rwr/pip.zip/pip/_internal/cli/main.py", line 60
    sys.stderr.write(f"ERROR: {exc}")
                                   ^
SyntaxError: invalid syntax
```

starbasessd, 2021-01-25:
Checking

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-01-25:
Using Ubuntu-20.04-desktop-amd64.iso for initial install, 4GB RAM, 20GB HD,
4 CPU.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-01-25:
I'll bet you ran a group of commands as a single copy & paste, rather than
as separate commands. I'll change the docs to indicate to NOT run as a
group copy/paste.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mbrivio, 2021-01-25:
not the case, just running these 3 commands on a fresh install give me the error:
```
apt-get install python2 -y
 curl https://bootstrap.pypa.io/get-pip.py --output get-pip.py
 python2 get-pip.py
```

I'm using an ubuntu server iso, 20.04, but it shouldn't matter

mbrivio, 2021-01-25:
There could be issues with the python2 package?

starbasessd, 2021-01-25:
That is what I'll be checking as soon as I get a base VM set up.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-01-25:
There was an issue with the upstream location of pip2 (they moved it) I
corrected the documentation. If the https:// path doesn't include 2.7, then
do a refresh on the page to get the corrected path.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-01-25:
Documentation updated. They (maintainers of pip2 install script) moved it. Check the instructions, and do a refresh on the page. The new curl path for the https:// site now has a 2.7 in the path.