motioneye-project/motioneye #1070: motion 4.2

donvipre, 2018-11-19:
I have a CentOS 7.5 machine running motioneye and motion 4.1 since a year now.

I started to build motion from here https://github.com/Motion-Project/motion
```bash
git clone
cd motion
autreconf -fiv
./configure
make
sudo make install
```

After now upgrading from 4.1 to 4.2 the live camera in motioneye does not work anymore.
I noticed also some errors in the log of my systemd unit motioneye
```
Nov 14 23:41:37 m1-server systemd[1]: Started motionEye Server.
Nov 14 23:41:37 m1-server systemd[1]: Starting motionEye Server...
Nov 14 23:41:37 m1-server meyectl[1012]: INFO: hello! this is motionEye server 0.39.3
Nov 14 23:41:38 m1-server meyectl[1012]: DEBUG: using motion version 4.2
Nov 14 23:41:38 m1-server meyectl[1012]: DEBUG: using ffmpeg version 2.8.15
Nov 14 23:41:38 m1-server meyectl[1012]: DEBUG: reading main config from file /etc/motioneye/motion.conf...
Nov 14 23:41:38 m1-server meyectl[1012]: DEBUG: loading additional config structure for main, without separators
Nov 14 23:41:38 m1-server meyectl[1012]: DEBUG: additional config section: network
Nov 14 23:41:38 m1-server meyectl[1012]: DEBUG: listing config dir /etc/motioneye...
Nov 14 23:41:38 m1-server meyectl[1012]: DEBUG: found camera with id 2
Nov 14 23:41:38 m1-server meyectl[1012]: DEBUG: reading camera config from /etc/motioneye/thread-2.conf...
Nov 14 23:41:38 m1-server meyectl[1012]: DEBUG: loading additional config structure for camera, without separators
Nov 14 23:41:38 m1-server meyectl[1012]: DEBUG: starting motion
Nov 14 23:41:38 m1-server meyectl[1012]: DEBUG: starting motion binary "/usr/local/bin/motion"
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [ALR] [ALL] conf_cmdparse: "webcontrol_html_output" replaced with "webcontrol_interface" after version 4.1.1
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [ALR] [ALL] conf_cmdparse: The "thread" option has been replaced by the "camera" after version 3.4.1
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file thread-2.conf
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [ALR] [ALL] conf_cmdparse: "ffmpeg_output_movies" replaced with "movie_output" after version 4.1.1
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [ALR] [ALL] conf_cmdparse: "quality" replaced with "picture_quality" after version 4.1.1
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [ALR] [ALL] conf_cmdparse: "output_pictures" replaced with "picture_output" after version 4.1.1
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [ALR] [ALL] conf_cmdparse: "ffmpeg_variable_bitrate" replaced with "movie_quality" after version 4.1.1
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [ALR] [ALL] conf_cmdparse: "ffmpeg_video_codec" replaced with "movie_codec" after version 4.1.1
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [ALR] [ALL] conf_cmdparse: "lightswitch" replaced with "lightswitch_percent" after version 4.1.1
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [ALR] [ALL] conf_cmdparse: "text_double" replaced with "text_scale" after version 4.1.1
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [ALR] [ALL] conf_cmdparse: "max_movie_time" replaced with "movie_max_time" after version 4.1.1
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [ALR] [ALL] conf_cmdparse: "ffmpeg_bps" replaced with "movie_bps" after version 4.1.1
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [ALR] [ALL] conf_cmdparse: "ffmpeg_output_debug_movies" replaced with "movie_output_motion" after version 4.1.1
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [ALR] [ALL] conf_cmdparse: "output_debug_pictures" replaced with "picture_output_motion" after version 4.1.1
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.2 Started
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (ALL)
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [INF] [ALL] conf_output_parms: Writing configuration parameters from all files (2):
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [INF] [ALL] Thread 0 - Config file: /etc/motioneye/motion.conf
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [INF] [ALL] daemon                    off
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [INF] [ALL] setup_mode                off
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [INF] [ALL] pid_file
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [INF] [ALL] log_file
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [INF] [ALL] log_level                 9
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [INF] [ALL] log_type                  ALL
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [INF] [ALL] quiet                     on
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [INF] [ALL] native_language           on
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [INF] [ALL] camera_name
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [INF] [ALL] camera_id                 0
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [INF] [ALL] target_dir
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [INF] [ALL] videodevice               /dev/video0
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [INF] [ALL] vid_control_params
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [INF] [ALL] v4l2_palette              17
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [INF] [ALL] input                     -1
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [INF] [ALL] norm                      0
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [INF] [ALL] frequency                 0
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [INF] [ALL] auto_brightness           0
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [INF] [ALL] tunerdevice
Nov 14 23:41:38 m1-server motion[1175]: [0:motion] [INF] [ALL] roundrobin_frames         1
[... 191 lines cut ...]
Nov 14 23:41:38 m1-server motion[1175]: [1:ml1] [INF] [NET] netcam_rtsp_set_parms: Setting up Normal resolution stream.
Nov 14 23:41:38 m1-server motion[1175]: [1:ml1] [INF] [NET] netcam_rtsp_set_path: Setting up rtsp via ffmpeg netcam
Nov 14 23:41:38 m1-server motion[1175]: [1:ml1] [INF] [NET] netcam_rtsp_set_rtsp: Normal resolution: Setting rtsp transport to tcp
Nov 14 23:41:38 m1-server motion[1175]: [1:ml1] [INF] [ENC] ffmpeg_avcodec_log: Failed to resolve hostname cam01.mac01.tuxsolutions.org: Der Name oder der Dienst ist nicht bekannt
Nov 14 23:41:38 m1-server motion[1175]: [1:ml1] [ERR] [NET] netcam_rtsp_open_context: Normal resolution: Unable to open camera((null)): Eingabe-/Ausgabefehler
Nov 14 23:41:38 m1-server motion[1175]: [1:ml1] [INF] [NET] netcam_rtsp_cleanup: Normal resolution: Shutting down network camera.
Nov 14 23:41:38 m1-server motion[1175]: [1:ml1] [NTC] [NET] netcam_rtsp_cleanup: Normal resolution: Shut down complete.
Nov 14 23:41:38 m1-server motion[1175]: [1:ml1] [ERR] [VID] vid_start: Netcam RTSP failed to open
Nov 14 23:41:38 m1-server motion[1175]: [1:ml1] [WRN] [ALL] motion_init: Could not fetch initial image from camera
Nov 14 23:41:38 m1-server motion[1175]: [1:ml1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
Nov 14 23:41:38 m1-server motion[1175]: [1:ml1] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 1 items
Nov 14 23:41:38 m1-server motion[1175]: [1:ml1] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 12 items
Nov 14 23:41:40 m1-server motion[1175]: [1:ml1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
Nov 14 23:41:40 m1-server motion[1175]: [1:ml1] [NTC] [VID] vid_start: Opening Netcam RTSP
Nov 14 23:41:40 m1-server motion[1175]: [1:ml1] [INF] [NET] netcam_rtsp_set_parms: Setting up Normal resolution stream.
Nov 14 23:41:40 m1-server motion[1175]: [1:ml1] [INF] [NET] netcam_rtsp_set_path: Setting up rtsp via ffmpeg netcam
Nov 14 23:41:40 m1-server motion[1175]: [1:ml1] [INF] [NET] netcam_rtsp_set_rtsp: Normal resolution: Setting rtsp transport to tcp
[... cut ...]
Nov 14 23:41:40 m1-server motion[1175]: [1:ml1] [NTC] [NET] netcam_rtsp_cleanup: Normal resolution: Shut down complete.
Nov 14 23:41:40 m1-server motion[1175]: [1:ml1] [ERR] [VID] vid_start: Netcam RTSP failed to open
Nov 14 23:41:40 m1-server motion[1175]: [1:ml1] [NTC] [ALL] motion_loop: Thread exiting
Nov 14 23:41:40 m1-server motion[1175]: [1:ml1] [DBG] [EVT] exec_command: Executing external command '/usr/lib/python2.7/site-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" stop 1'
```

ccrisan, 2018-11-20:
motionEye does not yet work with motion 4.2

donvipre, 2018-11-20:
ok, I'll be patient. Thank you for the fast response :)

deece, 2018-12-02:
I'm not sure this is related to motioneye - that log mentions DNS resolution failures.

donvipre, 2018-12-08:
so this could be related to motion 4.2 and DNS? ;) As the hostname is existing and resolvable on the machine. Thanks very much for the hint.

deece, 2018-12-08:
That's my thoughts, those log messages look like they originate from motion

electron1979, 2018-12-10:
> motionEye does not yet work with motion 4.2
Oh, that's why I am having issues!

I can only see one (local) motion cam out of 6 (3 on the Pi3 running motionOS and 3 on an Orange Pi PC+).
Motion does not show any of the PC+ cams until I pkill motion on the local Pi3 running motionOS, weird?
One of the "dependencies" was not installed before I had motion running (before I installed motionOS) and I'm using "daemon OFF" for motion.conf, because "ON" doesn't work for me (I'm using systemctl to manage motion, but have "daemon ON" for motionOS (as per [this](https://youtu.be/L9_imMScblE) guide).
I'm using the latest motion Debian repo and the October 2018 RPi desktop OS.

Very promising application! ;)

ttimasdf, 2018-12-22:
use `gdebi` to automatically install missing motion 4.2 dependency `libmicrohttpd12`. It works for me but not helps much. motion 4.2 deprecated a lot of configuration parameter names and generates a lot of warnings. It also remove `stream_motion` which slows down the stream when no motion detected. In a word it breaks things.

ccrisan, 2019-03-07:
https://github.com/ccrisan/motioneye/releases/tag/0.40rc1 adds support for Motion 4.2.

closed 2019-03-07 by ccrisan