motioneye-project/motioneye #1076: Please Add Mosquitto Client to the Docker Image

juan11perez, 2018-11-30:
thank you for this great package. 
Please consider Adding Mosquitto Client to the Docker Image

kleini, 2018-11-30:
Please describe the use case for having a Mosquitto client in the Docker image. Maybe it makes more sense to implement MQTT client features into motionEye.

juan11perez, 2018-11-30:
@kleini 
Thank you for your quick reply. motioneye is used by several home assistant users.
We're using mqtt in the motion_notification>command box to alert motion. 
I have manually installed the client in the docker container and works great. It'll off-course disappear if the container is replaced for whatever reason.

deece, 2018-12-01:
I'm currently working on adding MQTT directly to motion, but it's going to a little while before I upstream that, as I have a few other things to finish off first.

In the meantime, the recommended way to interact with MQTT is to call it as a notification command. I've just sent PR #1077 to support this.

juan11perez, 2018-12-01:
thank you for considering this request and the information.

JamBalaya56562, 2026-11-09:
Closing as a duplicate of #877, which asks for the same. Please follow and vote there.