motioneye-project/motioneye #1079: Raspberry Pi 3 with Motioneye OS not Detecting Pi Camera

wenzelp1, 2018-12-01:
Hi All- I've got a Raspberry Pi 3 with the Pi Camera v2.  I'm running MotionEye OS but after it configures and I login to the web dashboard the camera is not displayed.  This may be a basic question but I've triple checked my work, reinstalled the OS just to be sure and checked to ensure the camera works properly on Raspbian.   Still no luck.  It's very frustrating as surely I'm missing something simple; can anyone help?

Thank you!!

[image]
[image]
[image]

ccrisan, 2018-12-03:
Does your camera work with Raspbian?

wenzelp1, 2018-12-03:
thanks for the follow-up, yes, I've confirmed it works on Raspbian

elmicha, 2018-12-04:
Did you disconnect the camera cable between using Raspbian and Motioneye OS? If so, did you make sure the orientation of the cable contacts is correct on both ends?

This might sound like I would suggest that you're an idiot, but in fact I have done exactly that a few days ago. Disconnected the cable, reconnected it and paid absolutely no attention to the orientation of the cable. It took over an hour for me to realize that oh, that cable has two sides, and only one of them has connectors.

wenzelp1, 2018-12-07:
thanks for the feedback elmicha, it's a very fair question to ask :-).  I have double checked that early on and it's correct.  Am I missing something in the setup?  Flash the SD card with motioneye os, fire up the Pi with the camera connected and an ethernet cable plugged in, let it configure / boot up, and then log into the motioneye dashboard from a seperate PC using the Pi's IP address?

wenzelp1, 2018-12-08:
bump

massively appreciate the help!

elmicha, 2018-12-08:
No, what you did sounds correct. You can't have the same camera as MMAL and V4L2 camera at the same time, but you probably already tried each on its own. If you enable ssh, login with ssh and type `dmesg`, I guess you can see that the camera is not recognized?

JamBalaya56562, 2026-11-08:
Closing, as the `dmesg` output asked for above never came, while the camera worked on Raspbian. This was on motionEyeOS, which is maintained in its own repository (motioneye-project/motioneyeos), so please report it there if it still happens.

closed 2026-11-08 by JamBalaya56562