motioneye-project/motioneye #1081: Timelapse generation stuck saying "A timelapse movie is already being created"

warlon, 2018-12-01:
Timelapse movie generation stopped working for some reason, perhaps after being started simultaneously from two accounts. How can I fix this situation? Are there some temporary files I can just clear out? I'm running the latest version on a raspberry pi 3b+.

numbatboy, 2019-01-19:
Exact same problem on raspi 3B+ on latest Raspian Stretch.

numbatboy, 2019-01-19:
$ systemctl restart motioneye 
worked for my installation

Matze1302, 2020-03-30:
Same problem here, over Website i’ve got message: „A timelapse movie is already beeing created“ but i did not create anything before. Some days earlier everything works fine and nothing changed.
Restart doesnt help.
i hope there is a such as a temporary file to delete to get it work again, any idea?
Over commandline, the timelapse creation is okay but my wife don‘t like to use cmdline for it ^^
Motioneyeos version: 20200203

thank you

Blackraz0r, 2020-12-16:
sry for necroing a thread but i have the same issue.

Matze, may you describe me how you generated the timelapse via ssh?

starbasessd, 2020-12-16:
@Blackraz0r 
What is your setup? At a minimum, which Pi and which version of motionEyeOS?
if 20200606, please try dev20201026 or 20100911

Blackraz0r, 2020-12-16:
i also managed to make the timelapse at 30fps but 20 or 60 are failing.

you are right, i have
20200606 currently.
ill update tonight and will report back if it still fails.

closed 2026-11-08 by Marijn0