motioneye-project/motioneye #1084: writeimage.sh does no run when i use the run command

hareyleyland, 2018-12-03:
summary : i have been trying to write motioneyeos to my SD card with macOSX but keep getting errors. 
i copied the line of code to the terminal  ./writeimage.sh -d /dev/mmcblk0 -i "/path/to/motioneyeos.img" and it said -bash: ./writeimage.sh: No such file or directory i have been doing some research into this but found nothing.
any help would be great thanks,
harvey leyland

elmicha, 2018-12-04:
Maybe you should just use [Etcher](https://etcher.io/) instead. It has a GUI, and if you don't need to set the WiFi password it should have no downsides.

If you really want to use the command line:

You are probably in the wrong directory when you started `./writeimage.sh`. Use the `cd` command to navigate to the directory where you downloaded the `./writeimage.sh`. 

Also don't use the filename "/path/to/motioneyeos.img" verbatim, it is a placeholder. If you downloaded `./writeimage.sh` and the `motioneyeos*.img.xz` file into the same directory, omit the path, but use the filename of the file that you downloaded, e.g.

    ./writeimage.sh -d /dev/mmcblk0 -i motioneyeos-raspberrypi-dev20181130.img.xz

I don't know anything about Macs, but chances are that your SD card reader is not /dev/mmcblk0. Maybe ask in a Mac forum how you can find the device name of your SD card reader. Ok, [apparently](https://gist.github.com/jordelver/3139365) you can use `diskutil list` to list your disks. Be 100% sure to select the correct disk, or you will overwrite one of your hard disks!

JamBalaya56562, 2026-11-08:
Closing as answered above: `writeimage.sh` is the motionEyeOS image writer, so it has to be run from the folder it was downloaded to, and a tool like Etcher writes the image without it. motionEyeOS is maintained in its own repository (motioneye-project/motioneyeos).

closed 2026-11-08 by JamBalaya56562