motioneye-project/motioneye #1085: Cannot access Embed URL of remote Motioneye Camera

judalvarezca, 2018-12-04:
Hello, I'm having problems on streaming with the embed URL on remote motioneye cameras.

My setup:
- Motioneye 0.39.2 running on Ubuntu 18.04,
- 3 Raspberry Pi Zero W, with raspbian stretch lite and motioneye 0.39.2,

I'm able to see each frame in the server, but when I try to access one of the frames from embed URL, I see this:

[image]

By looking at the log file, I'm seeing this error:
```
 DEBUG: getting config for remote camera 1 on http://192.168.1.221:9876/config/1
   ERROR: failed to create root directory "/home/pi/motioneye/media/Camera1": [Errno 13] Permission denied: '/home/pi'
Traceback (most recent call last):
  File "/home/judalvarezca/.virtualenvs/mymotioneye/local/lib/python2.7/site-packages/motioneye/config.py", line 903, in motion_camera_ui_to_dict
    os.makedirs(data['target_dir'])
  File "/home/judalvarezca/.virtualenvs/mymotioneye/lib/python2.7/os.py", line 150, in makedirs
    makedirs(head, mode)
  File "/home/judalvarezca/.virtualenvs/mymotioneye/lib/python2.7/os.py", line 150, in makedirs
    makedirs(head, mode)
  File "/home/judalvarezca/.virtualenvs/mymotioneye/lib/python2.7/os.py", line 150, in makedirs
    makedirs(head, mode)
  File "/home/judalvarezca/.virtualenvs/mymotioneye/lib/python2.7/os.py", line 157, in makedirs
    mkdir(name, mode)
OSError: [Errno 13] Permission denied: '/home/pi'
   DEBUG: 200 GET /picture/4/frame/ (127.0.0.1) 423.98ms
   ERROR: HTTP 404: Not Found (not found)
 WARNING: 404 GET /picture/4/frame/static/css/jquery.timepicker.css?v=0.39.2 (127.0.0.1) 0.68ms
   ERROR: HTTP 404: Not Found (not found)
 WARNING: 404 GET /picture/4/frame/static/css/ui.css?v=0.39.2 (127.0.0.1) 0.95ms
   ERROR: HTTP 404: Not Found (not found)
 WARNING: 404 GET /picture/4/frame/static/css/main.css?v=0.39.2 (127.0.0.1) 0.69ms
   ERROR: HTTP 404: Not Found (not found)
 WARNING: 404 GET /picture/4/frame/static/css/frame.css?v=0.39.2 (127.0.0.1) 0.74ms
   ERROR: HTTP 404: Not Found (not found)
 WARNING: 404 GET /picture/4/frame/static/js/css-browser-selector.js?v=0.39.2 (127.0.0.1) 0.86ms
   ERROR: HTTP 404: Not Found (not found)
 WARNING: 404 GET /picture/4/frame/static/js/main.js?v=0.39.2 (127.0.0.1) 0.88ms
   ERROR: HTTP 404: Not Found (not found)
 WARNING: 404 GET /picture/4/frame/static/js/jquery.timepicker.min.js?v=0.39.2 (127.0.0.1) 0.82ms
   ERROR: HTTP 404: Not Found (not found)
 WARNING: 404 GET /picture/4/frame/static/js/jquery.mousewheel.js?v=0.39.2 (127.0.0.1) 0.75ms
   ERROR: HTTP 404: Not Found (not found)
 WARNING: 404 GET /picture/4/frame/static/js/ui.js?v=0.39.2 (127.0.0.1) 5.69ms
   ERROR: HTTP 404: Not Found (not found)
 WARNING: 404 GET /picture/4/frame/static/js/jquery.min.js?v=0.39.2 (127.0.0.1) 1.36ms
   ERROR: HTTP 404: Not Found (not found)
 WARNING: 404 GET /picture/4/frame/static/js/frame.js?v=0.39.2 (127.0.0.1) 0.82ms
   ERROR: HTTP 404: Not Found (not found)
 WARNING: 404 GET /picture/4/frame/static/img/no-camera.svg (127.0.0.1) 0.87ms
   ERROR: HTTP 404: Not Found (not found)
 WARNING: 404 GET /picture/4/frame/static/js/jquery.timepicker.min.js?v=0.39.2 (127.0.0.1) 0.83ms
   ERROR: HTTP 404: Not Found (not found)
 WARNING: 404 GET /picture/4/frame/static/js/jquery.mousewheel.js?v=0.39.2 (127.0.0.1) 0.58ms
   ERROR: HTTP 404: Not Found (not found)
 WARNING: 404 GET /picture/4/frame/static/js/ui.js?v=0.39.2 (127.0.0.1) 0.69ms
   ERROR: HTTP 404: Not Found (not found)
 WARNING: 404 GET /picture/4/frame/static/js/main.js?v=0.39.2 (127.0.0.1) 1.21ms
   ERROR: HTTP 404: Not Found (not found)
 WARNING: 404 GET /picture/4/frame/static/js/frame.js?v=0.39.2 (127.0.0.1) 0.59ms
   ERROR: HTTP 404: Not Found (not found)
 WARNING: 404 GET /picture/4/frame/static/img/no-camera.svg (127.0.0.1) 1.45ms
   ERROR: HTTP 404: Not Found (not found)
 WARNING: 404 GET /picture/4/frame/static/favicon.ico?v=0.39.2 (127.0.0.1) 0.96ms
   ERROR: HTTP 404: Not Found (not found)
 WARNING: 404 GET /picture/4/frame/static/favicon.ico?v=0.39.2 (127.0.0.1) 0.92ms
```
I understand this is related to the mkdir on the Raspberries, but what can I do to solve this?

Thank you in advance.

judalvarezca, 2018-12-05:
Well, Today I started reinstalling everything from scratch. I reinstalled raspbian+motioneye on one of my cameras again and made another virtual environment with a raw new installation. I'm now running motioneye 0.39.3, with raspbian stretch and Ubuntu Bionic. The problem is the same, but I have some new information. When I include the MMAL camera on the motioneye server running locally on my raspberry, the system finds it normally and I can access the embed URL (inside the local WiFi network):

[image]

However, when I include it as a Remote motioneye camera in my server (which is the bridge between two networks) it shows perfectly on it. I need to feed that frame to another webpage, so I was planning to use the embed URL, but I cannot access the Remote motioneye camera with the error I mentioned previously. 

Testing the waters, I tried to add a simple MJPEG camera from the streaming of the local motioneye on the raspberry, and it worked. But I need to know if I'm making a mistake trying to access to the frame from the remote motioneye camera inside my Motioneye server, or if there is a workaround, since I'm having 2 of the same camera for the streaming purposes, and one of them have no motion detection at all. 
[image]
[image]

Sorry about my english, still learning.

Regards, Juan.

bahamut657, 2019-03-24:
I'm also experiencing this issue..
My remote cameras (installing motioneye) are configured usigng local V4L driver
When trying to stream from Embed URL, camera1 is visible correctly, camera2 shows broken image symbol.
If accessing URL directly to not showed camera, it seems to work (example http://camera2/picture/1/frame works, http://camgateway/picture/2/frame doesn't)

JamBalaya56562, 2026-11-08:
Closing, as this was with motionEye 0.39, and there were no further replies. The error in the log comes from the hub trying to create `/home/pi/motioneye/media/Camera1`, the remote camera's own folder, which does not exist on the hub. Remote cameras are reached differently since 0.44.0 (#3332), which replaced the old URL signatures. Please open a new issue with the log if the embed URL of a remote camera still fails with current releases.

closed 2026-11-08 by JamBalaya56562