motioneye-project/motioneye #1088: Motioneye is not recognizing PS3 Eye camera

Amarok73, 2018-12-06:
Hello, 
Since a while, my MotionEye can't recognize camera. The command v4l2-ctl --list-devices returns information that camera is found, but in the MotionEye it can't be recognized.

[image]

[image]

Amarok73, 2018-12-06:
Little update, after deleting the camera, it appears in "Add Camera" dialogue form and I can add it, but then "meyectl startserver" continuously keep showing the error about "mjpg client for camera 1 on port 8081", while the only process hooked to this port is the "motion" apparently started by the "meyectl startserver" itself.

ccrisan, 2018-12-15:
@Amarok73 does your camera work with:

1. other V4L2 programs?
2. standalone Motion?

Amarok73, 2018-12-16:
Sorry for being such indolent ignorant, but I am not sure, how could I verify it...?
The message about "connection problem detected" keeps appearing in log only when I am trying to access view from camera on the MotionEye interface. Otherwise, no messages appears there.

Amarok73, 2018-12-24:
It took some time, but I have narrowed down symptoms a bit. It seems, that only preview from camera in the MotionEye web interface is not working properly, while all captures are done correctly according to setting. Still I'd be grateful if someone could help me to get MotionEye back to full functionality...

Amarok73, 2019-01-31:
Ok, found time to get back to the issue, and standalone Motion cause to show only gray screen with some text overlay on it, as on the screenshot...
[image]

Amarok73, 2019-01-31:
Another update, there were no video0 in /dev/, so according to [this post](https://www.linuxquestions.org/questions/linux-newbie-8/missing-dev-video0-261133/#post1378255) i've created this object, and there were still gray picture, then i replug the camera into USB port and now i can see the propper picture. 

Last lines in motion.log before i've reconnected camera was as follows:
[image]

And picture from the camera...
[image]

Going back to the MotionEye, still no preview, and trying to start server manually by "meyectrl startserver" shows information that "Address is already in use". Trying to kill the processes containing "motion" in the name seems to have no real effect, as motioneye is starting automatically almost immediately. In the /var/log/motioneye there's no any file that could contain log.

I think i am at the point, where I really need some advices...

Amarok73, 2019-03-24:
Kind of sorted it out. 
It seems, that change of host name caused these problems. Reverted to old name from the moment when installed the MotionEye, and now I can see the picture in Control Panel.
Question is then, what should I change in MotionEye configuration, after host name change?

Fenestrane1, 2024-06-14:
> Kind of sorted it out. It seems, that change of host name caused these problems. Reverted to old name from the moment when installed the MotionEye, and now I can see the picture in Control Panel. Question is then, what should I change in MotionEye configuration, after host name change? @ccrisan Calin, may I count on You on this one?

Hi,
i think im having same kind of problem, having that gray screen saying unable to open video device. Just raised this issue today https://github.com/motioneye-project/motioneye/issues/3009, would be glad if you can help. seeing you overcame this issue. Did your camera icon also show the black thumbnail with a no camera image on it sometimes. Mine is sometimes alternating between that and unable to open video device

MichaIng, 2024-06-14:
I'm closing this ancient issue, which is also not related to RPi camera modules, but regular V4L2 ones.

closed 2024-06-14 by MichaIng