motioneye-project/motioneye #1089: Motion Eye not saving Video

shahzadgodil, 2018-12-11:
I have an IP camera setup. But MotionEye is not recording video in their specified path

Also when I am trying to play it in Preview icon, it is showing me the error "No Media File"

Screenshot of Error
https://www.awesomescreenshot.com/image/3782391/d4878fd9dd8cb01dc67c421cd18ab772

Path:
/var/lib/motioneye/Camera1

Please resolve 

Thanks
Shahzad Godil

grphx7, 2019-01-12:
Make sure your system has the proper codec installed for the filetype you're trying to install.

joshua-animas, 2020-01-19:
Had the same issue. Switched the format to .mp4 (H.264) and it did the trick. Good call.

JamBalaya56562, 2026-11-07:
Closing, as switching the movie format to H.264 (.mp4) fixed the same problem for another user above. Please open a new issue with your motionEye log if movies are still not saved.

closed 2026-11-07 by JamBalaya56562